RavenSystem/esp-homekit-devices #1129: Add support for AС (air conditioner)

NikDevx, 2020-09-10:
Hello, @RavenSystem!
Thanks you for this amazing work!
Will you be adding AC support in the future?

hillaliy, 2020-09-10:
Hi HAA support AC.
see wiki, thermostat 25

NikDevx, 2020-09-11:
> Hi HAA support AC.
> see wiki, thermostat 25

Hi 
Homekit has different interface and device type fo AC

https://developer.apple.com/homekit/specification/

RavenSystem, 2020-09-11:
There is not any new HomeKit interface for AC since summer 2019.

NikDevx, 2020-09-11:
> There is not any new HomeKit interface for AC since summer 2019.

Does the thermostat and AC have the same function?

pro19hsu, 2020-09-14:
> 自2019年夏季以來，AC沒有任何新的HomeKit接口。

The working mode of HAA Infra Red is different from that of my air-conditioning remote control Infra Red.
My air-conditioning remote will send Config at once.
Is there a corresponding setting method?

pro19hsu, 2020-09-14:
> There is not any new HomeKit interface for AC since summer 2019.

The working mode of HAA Infra Red is different from that of general air-conditioning remote control Infra Red.
Generally, the air-conditioning remote control will send Config once.
Is there a corresponding setting method?

pro19hsu, 2020-09-14:
> There is not any new HomeKit interface for AC since summer 2019.

@RavenSystem 
The working mode of HAA Infra Red is different from that of general air-conditioning remote control Infra Red.
Generally, the air-conditioning remote control will send Config once.
Is there a corresponding setting method?

RavenSystem, 2020-09-14:
I don't know how General Electric AC works. HAA has been tested with Mitsubishi, Daikin, Fujitsu, Samsung and others similar AC managed by IR. If General AC works with a special system sending config only once and does not allow to send full commands each time, it can be difficult to manage with HAA. You can send several commands at same time, with a custom delay, it could work with General AC, but you must try it.

closed 2020-09-14 by RavenSystem