RavenSystem/esp-homekit-devices #1135: 220V switch question

outoforder20, 2020-09-14:
I tried to modify adv_button lib but i could not make it.

Here some sources for 220V switch:

https://templates.blakadder.com/moes-MS-104B.html

https://tasmota.github.io/docs/Buttons-and-Switches/

https://community.home-assistant.io/t/moes-ms104-b-howto-config/172237

https://templates.blakadder.com/blitzwolf_BW-SS5_2.html

I think it is possible with right code

Latest adv_button lib didn't work (throws exceptions) so i tried adv_button 3.1.0 
https://github.com/RavenSystem/esp-homekit-devices/commit/c85116e20ed30be17c175c9579b8bbfdc49ee616#diff-65c44ec7ced1ab18cbebc176854ba497

Here it turns on-off sometimes but not stabile. Trying with momentry and toggle options

main.c
````
       //adv_button_register_callback_fn(button_gpio_1, button_simple1_intr_callback, 0, NULL, 0);
        adv_button_register_callback_fn(button_gpio_1, button_simple1_intr_callback, 1, NULL, 0);
        //adv_button_register_callback_fn(button_gpio_2, button_simple2_intr_callback, 0, NULL, 0);
        adv_button_register_callback_fn(button_gpio_2, button_simple2_intr_callback, 1, NULL, 0);
````

adv_button.c
````
#define ADV_BUTTON_DEFAULT_EVAL     (1)

#define DOUBLEPRESS_TIME            (400)

#define LONGPRESS_TIME              (DOUBLEPRESS_TIME + 10)
#define VERYLONGPRESS_TIME          (1500)
#define HOLDPRESS_TIME              (8000)

#define BUTTON_EVAL_DELAY_MIN       (10)
#define BUTTON_EVAL_DELAY_DEFAULT   (350)
#define BUTTON_EVAL_DELAY_MAX       (BUTTON_EVAL_DELAY_MIN + 245) //245

#define DISABLE_PRESS_COUNT         (131)

#define DISABLE_TIME                (ADV_BUTTON_DEFAULT_EVAL * 350)
#define MIN(x, y)                   (((x) < (y)) ? (x) : (y))
#define MAX(x, y)                   (((x) > (y)) ? (x) : (y))
````

RavenSystem, 2020-09-14:
You don't need to modify anything. Adv_button_library works as is, and it is used by HAA without any modification.

outoforder20, 2020-09-14:
@RavenSystem thank you for fast reply.

BW-SS5 and MS-104B have 220V AC input for Switch1 and Switch2.. There is no response from the relays, 
They are not like 3.3v dc inputs.

closed 2020-10-14 by stale[bot]