RavenSystem/esp-homekit-devices #1138: How turn a light on while the thermostat is heating?

tlagarto, 2020-09-18:
I have a Sonoff TH10 working as a thermostat with humidity sensor (t:25).
I would like to switch on another device (Sonoff mini) while the thermostat is on and is heating from current temperature to goal temperature.
Once it reaches the goal temperature and the TH10 stops heating, the mini should be switched off too.

I've checked the HomeKit app, but it seems I have no option to "thermostat is heating", did I missed something?

I was thinking if setting a notification ("m") on the thermostat to a virtual switch (on the TH10 too).
Then, I can sync that virtual switch with the Sonoff mini on the HomeKit App.
But I can't figure out how to set up the notification when the thermostat is on AND heating... in other words, every time the relay is active, I want the virtual switch to be "on"...

Any help on how to do this?

tlagarto, 2020-09-19:
Got it.
Here it is, in case anyone wants to use:
`{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}],"m":[{"g":2,"v":0}]},"1":{"a":0},"2":{"a":0},"3":{"r":[{"g":12,"v":1}],"m":[{"g":2,"v":1}]},"4":{"a":3},"5":{"a":0},"t":25,"s":5,"w":3,"n":4,"j":60,"g":14},{"b":[{"g":0,"t":1}],"s":0},{"1":{"s":[{"a":1}]},"i":0.5,"b":[],"s":0},{"1":{"s":[{"a":2}]},"t":2,"i":0.5,"b":[],"s":0}]}`

wayn3, 2020-09-19:
I think you should do this inside Homekit, however you would need to do it in a third party app like eve or home+

closed 2020-10-19 by stale[bot]