RavenSystem/esp-homekit-devices #1139: Sonoff Basic R2 with Haa 3.1.2

huta76, 2020-09-19:
I'm new to HAA and have 2 issues:

1.  Pairing is really difficult.  After a lot of times, it happens but takes a number of times of performing the same thing.  Is this normal?
2. I notice the switch on Home keeps losing connectivity and establishing it again.  I have a couple of switches on standard Sonoff firmware paired through Homebridge and this never happens.  Ultimate goal is to eliminate homebridge.   Any settings that can be tinkered with please?

Excellent system otherwise! Keep up the good work!

ondratan, 2020-09-25:
Hi, everything has already been solved in closed issues.
First of all verify that the installation was completed correctly, you can capture log with this command `nc -kulnw0 45678`.
If you want to add the device again, you have to reset HomeKit ID in device. Regarding devices that do not responding try to use Merlin firmware in your router, it helped to me.

huta76, 2020-10-01:
Thank you for your reply. I have since upgraded to latest 3.4.2 but unresponsiveness persists.    I have since done a bit more troubleshooting, and I can still ping the device so, wifi connectivity is not the issue.   If I press a button on the switch, it somehow "clicks" it back into connectivity.   I am currently using the Arris Cable modem supplied by provider.  Will look into this Merlin firmware though.  Thanks for your input.

ondratan, 2020-10-01:
Have a look here https://github.com/RavenSystem/esp-homekit-devices/issues/130

huta76, 2020-10-01:
Thanks again Ondratan... that was really helpful.  I _think_ it was my LG OLED.   It was also stuck on No response most of the time.  I removed it from the Homekit config.   I also restarted the router and it seems to be stable.  I will continue testing and update.   I think this should be part of the wiki and would be more than happy to contribute and document this for other newbies not to be turned off by this thinking that it is unstable

huta76, 2020-10-07:
I can confirm all is good and stable.  Thanks for all your help

closed 2020-10-07 by huta76