RavenSystem/esp-homekit-devices #1143: Problems with OTA update

ptath, 2020-09-20:
MagicHome LED, updating from 3.0.0, tried many times. Updating stops and device boots normally on 3.0.0

logs:

```
Advanced ESP Logger (c) 2020 José Antonio Jiménez Campos

0006.841 192.168.1.191-5C63EE: 
0006.843 192.168.1.191-5C63EE: *** STARTING UPDATE PROCESS
0006.847 192.168.1.191-5C63EE: Running OTAMAIN
0006.853 192.168.1.191-5C63EE: -----
0007.840 192.168.1.191-5C63EE: NEW CONNECTION LocalPort=e31a DNS..OK IP addr: 140.82.121.4 Local..OK Remote..no buf for action frame
...
0009.711 192.168.1.191-5C63EE: OK SSL..OK set_fd to github.com port 443..no buf for action frame
0009.713 192.168.1.191-5C63EE: no buf for action frame
0009.714 192.168.1.191-5C63EE: no buf for action frame
0011.028 192.168.1.191-5C63EE: OK
0011.033 192.168.1.191-5C63EE: sent OK
0011.196 192.168.1.191-5C63EE: ret = 1024
0011.198 192.168.1.191-5C63EE: 
0011.199 192.168.1.191-5C63EE: HTTP/1.1 302 Found
0011.200 192.168.1.191-5C63EE: date: Sun, 20 Sep 2020 17:43:11 GMT
0011.238 192.168.1.191-5C63EE: content-security-policy: default-src 'none'; base-uri 'self'; block-all-mixed-content; connect-src 'self' uploads.github.com www0013.074 192.168.1.191-5C63EE: NEW CONNECTION LocalPort=e31b DNS..OK IP addr: 140.82.121.4 Local..OK Remote..no buf for action frame
0015.649 192.168.1.191-5C63EE: GET /166993980/4153df00-fabd-11ea-990a-61053970aaca?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20200920%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20200920T174312Z&X-Amz-Expires=300&X-Amz-Signature=ff61178b0c56a83eadb101b508be=attachment%3B%20filename%3Dotaversion&response-content-type=application%2Foctet-stream HTTP/1.1
0016.422 192.168.1.191-5C63EE: NEW CONNECTION LocalPort=e31c DNS..OK IP addr: 52.216.137.52 Local..OK Remote..OK SSL..OK set_fd to github-production-release-asset-2e65be.s3.amazonaws.com port 443..no buf for action frame
0021.053 192.168.1.191-5C63EE: OK
0021.062 192.168.1.191-5C63EE: sent OK
0021.286 192.168.1.191-5C63EE: ret = 464
0021.288 192.168.1.191-5C63EE: 
0021.289 192.168.1.191-5C63EE: HTTP/1.1 206 Partial Content
0021.290 192.168.1.191-5C63EE: x-amz-id-2: 8PfxJlhq7vZEL0gKufHwy5ejiwXjWuoxp10rCK6t1p8tGtldJep3wziZKCRou89rv8s+4UIYiN0=
0021.292 192.168.1.191-5C63EE: x-amz-request-id: E035ED5E9CAD209B
0021.312 192.168.1.191-5C63EE: Content-Type: application/octet-stream
0021.316 192.168.1.191-5C63EE: Content-Length: 2
312Z&X-Amz-Expires=300&X-Amz-Signature=ff61178b0c56a83eadb101b508bed090a12c93d98aebea09c8d0a9304fca96c7&X-Amz-SignedHeaders=host0022.456 192.168.1.191-5C63EE: NEW CONNECTION LocalPort=e31d DNS..OK IP addr: 52.216.163.27 Local..OK Remote..OK SSL..OK set_fd to github-production-release-asset-2e65be.s3.amazonaws.com port 443..no buf for action frame
0027.624 192.168.1.191-5C63EE: VERSION of otaversion: 4.1.1
0027.625 192.168.1.191-5C63EE: Get sign
0029.581 192.168.1.191-5C63EE: NEW CONNECTION LocalPort=e31e DNS..OK IP addr: 140.82.121.4 Local..OK Remote..OK SSL..OK set_fd to github.com port 443..OK
0029.587 192.168.1.191-5C63EE: sent OK
0029.841 192.168.1.191-5C63EE: ret = 1024
0029.843 192.168.1.191-5C63EE: 
0029.845 192.168.1.191-5C63EE: HTTP/1.1 302 Found
0029.847 192.168.1.191-5C63EE: date: Sun, 20 Sep 2020 17:44:15 GMT
.githubstatus.com collector.githubapp.com api.github.com www.google-analytics.com github-cloud.s3.amazonaws.com github-production-repository-file-5c1aeb.s3.amazonaws.com github-production-upload-manifest-file-7fdce7.s3.amazonaws.com github-production-user-asset-6210df.s3.amazonaws.com cdn.optimizely.com logx.optimizely.com
0029.924 192.168.1.191-5C63EE: 
0030.136 192.168.1.191-5C63EE: NEW CONNECTION LocalPort=e31f DNS..OK IP addr: 140.82.121.4 Local..OK Remote..OK SSL..OK set_fd to github.com port 443..no buf f0030.139 192.168.1.191-5C63EE: no buf for action frame
0031.334 192.168.1.191-5C63EE: OK
0031.340 192.168.1.191-5C63EE: sent OK
0031.741 192.168.1.191-5C63EE: ret = 1024
0031.743 192.168.1.191-5C63EE: 
0031.744 192.168.1.191-5C63EE: HTTP/1.1 302 Found
0031.745 192.168.1.191-5C63EE: date: Sun, 20 Sep 2020 17:44:17 GMT
0031.746 192.168.1.191-5C63EE: content-type: text/html; charset=utf-8
0031.752 192.168.1.191-5C63EE: location: https://github-production-release-asset-2e65be.s3.amazonaws.com/166993980/3f8a1b80-fabd-11ea-905e-cb27edf38b08?X-Amz-Ast&actor_id=0&key_id=0&repo_id=166993980&response-content-disposition=attachment%3B%20filename%3Dhaaboot.bin.sec&response-contenor_id=0&key_id=0&repo_id=166993980&response-content-disposition=attachment%3B%20filename%3Dhaaboot.bin.sec&response-content-type=application%2Foctet-stream
=attachment%3B%20filename%3Dhaaboot.bin.sec&response-content-type=application%2Foctet-stream HTTP/1.1
0032.425 192.168.1.191-5C63EE: NEW CONNECTION LocalPort=e320 DNS..OK IP addr: 52.217.14.244 Local..OK Remote..OK SSL..OK set_fd to github-production-release-asset-2e65be.s3.amazonaws.com port 443..no buf for action frame
0037.021 192.168.1.191-5C63EE: OK
0037.029 192.168.1.191-5C63EE: sent OK
0037.229 192.168.1.191-5C63EE: ret = 471
0037.231 192.168.1.191-5C63EE: 
0037.232 192.168.1.191-5C63EE: HTTP/1.1 206 Partial Content
0037.233 192.168.1.191-5C63EE: x-amz-id-2: /4QF4QWT/biUNN9rkJmyvnJ5GcVHdU/aMsK1t9LQUmUHr18DrsBlxPXDxGAcABK1Y7srp/gbQv0=
0037.238 192.168.1.191-5C63EE: Last-Modified: Sat, 19 Sep 2020 19:15:32 GMT
0037.243 192.168.1.191-5C63EE: ETag: "90520468c8f9e4642510a37901aa9733"
0037.260 192.168.1.191-5C63EE: Server: AmazonS3
0037.262 192.168.1.191-5C63EE: 
417Z&X-Amz-Expires=300&X-Amz-Signature=b63adc303e60cc247a5f9b68b7ab496180034cd77c6a8a2da52c6bc040cb2701&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=166993980&response-content-disposition=attachment%3B%20filename%3Dhaaboot.bin.sec&response-content-0038.429 192.168.1.191-5C63EE: NEW CONNECTION LocalPort=e321 DNS..OK IP addr: 52.216.179.99 Local..OK Remote..OK SSL..OK set_fd to github-production-release-asset-2e65be.s3.amazonaws.com port 443..no buf for action frame
0043.079 192.168.1.191-5C63EE: OK
0043.294 192.168.1.191-5C63EE:  Downloaded 104 Bytes
```

shanghaix, 2020-09-20:
I have the exact same issue, I am not able to install new devices because they cannot get the OTA update.

ptath, 2020-09-20:
3.3.1 installed perfectly

closed 2020-09-20 by ptath