RavenSystem/esp-homekit-devices #1144: Please examples for doorbell

toshibochan, 2020-09-20:
Anyone have examples for push button for doorbell config?

toshibochan, 2020-09-20:
Doorbell button can trigger camera? I want doorbell send existing homekit security cameras as doorbell.

shanghaix, 2020-09-20:
Yes you can do that, using the homebridge camera plugin, when someone presses on a switch that you can define it will send a notification with a capture from your camera.

bj-tracer, 2020-09-21:
"This accessory works like a binary sensor, and it only sends a notification when status 0 is reached."
It sends notification like a door or window switch, I think.

toshibochan, 2020-09-22:
I have homekit compatible camera so I no need plugin for the camera. I just need the button to trigger the notification

bj-tracer, 2020-09-23:
> I have homekit compatible camera so I no need plugin for the camera. I just need the button to trigger the notification

Try to place cam and button to the same room at the Home app.

toshibochan, 2020-09-23:
I did but not working

pilot1981, 2020-09-23:
it's possibile to have an example of HAA JSON of doorbell without using homebridge??

ooomakon, 2020-09-23:
{"c":{"b":[{"g":0,"t":5}],"l":13},"a":[{"t":13,"s":0,"f1":[{"g":14,"t":1}],"i":2}]}

RavenSystem, 2020-09-23:
Doorbell example accessory, with button connected to GPIO 14:

`{"t":13,"f0":[{"g":14}]}`

toshibochan, 2020-09-23:
Thanks I will try later. I need to buy more esp8266.

pilot1981, 2020-09-26:
So it’s better to add a dummy switch to enter easily in setup mode?!

bj-tracer, 2020-09-29:
Yesterday test this.
It can't to activate any camera because HK see it like interkom or camera.
If you try to change a settings for this device at HK then app faulting.
It sends notification only and nothing else.

closed 2020-10-29 by stale[bot]

lukasa1993, 2021-01-07:
@RavenSystem any update on this ? it seems ios doenst accept button as doorbell camera is required