RavenSystem/esp-homekit-devices #1145: Problem connecting on configuration page on 331

sorriso93, 2020-09-21:
Hello no way to access setup page once flashed, I'm using wemos D1 mini as usual
**On Mac:**

Home Accessory Architect v3.3.1
Developed by José Antonio Jiménez Campos (@RavenSystem)

SETUP MODE
WiFi config init
No WiFi config found
Start AP SSID=HAA-1D83B4
Start DHCP server
mode : sta(dc:4f:22:1d:83:b4) + softAP(de:4f:22:1d:83:b4)
add if0
add if1
bcn 100
Start DNS server
Start HTTP server
Start WiFi scan
scandone
add 1
aid 1
station: 78:4f:43:50:9d:f7 join, AID = 1
Client disconnected
rtos assert /Users/jose/Documents/Desarrollo/ESP8266-devel/esp-homekit-devices/sdk/esp-open-rtos/FreeRTOS/Source/timers.c 386
abort() invoked at 0x40204973.

Stack: SP=0x3fff6fd0
0x3fff6fd0: 00000000 00000003 3ffef778 40204976
0x3fff6fe0: 00000000 00000000 3fff0988 40204679
0x3fff6ff0: 00000000 00000000 3fff332c 00000000
0x3fff7000: 00000000 00000000 3fff2e90 0000005f
0x3fff7010: 3fff7258 00000001 3ffef698 0000000a
0x3fff7020: 3fff71be 3fff1878 3ffef778 4022455f
0x3fff7030: 3fff70b0 3fff70b0 40231e90 40205922
0x3fff7040: 0000005f 3fff7070 40231e90 40205b10

Free Heap: 21128
_heap_start 0x3fff2988 brk 0x3fffb864 supervisor sp 0x40000000 sp-brk 18332 bytes
arena (total_size) 36572 fordblks (free_size) 2796 uordblocks (used_size) 33776

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 2292, room 16 
tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4 
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   DOUT
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 0.
pp_task_hdl : 3ffeffe0, prio:14, stack:512
pm_task_hdl : 3ffef858, prio:1, stack:176
frc2_timer_task_hdl:0x3fff5108, prio:12, stack:200

**on iphone (IOS14):**

Home Accessory Architect v3.3.1
Developed by José Antonio Jiménez Campos (@RavenSystem)

SETUP MODE
WiFi config init
No WiFi config found
Start AP SSID=HAA-1D83B4
Start DHCP server
mode : sta(dc:4f:22:1d:83:b4) + softAP(de:4f:22:1d:83:b4)
add if0
add if1
bcn 100
Start DNS server
Start HTTP server
Start WiFi scan
scandone
add 1
aid 1
station: e6:68:5b:99:06:3b join, AID = 1
scandone
! Unknown: GET /hotspot-detect.html
Unknown
Redirecting to http://192.168.4.1/settings
Client disconnected
rtos assert /Users/jose/Documents/Desarrollo/ESP8266-devel/esp-homekit-devices/sdk/esp-open-rtos/FreeRTOS/Source/timers.c 386
abort() invoked at 0x40204973.

Stack: SP=0x3fff6fd0
0x3fff6fd0: 00000000 00000003 3fff2bb0 40204976
0x3fff6fe0: 00000000 00000000 3fff0988 40204679
0x3fff6ff0: 00000000 00000000 3fff31bc 00000000
0x3fff7000: 00000000 00000000 3fff2dc8 00000071
0x3fff7010: 3fff7258 00000001 3ffef698 0000000a
0x3fff7020: 3fff71d0 3fff1878 3fff2bb0 4022455f
0x3fff7030: 3fff70b0 3fff70b0 40231e90 40205922
0x3fff7040: 00000071 3fff7070 40231e90 40205b10

Free Heap: 21004
_heap_start 0x3fff2988 brk 0x3fffbe90 supervisor sp 0x40000000 sp-brk 16752 bytes
arena (total_size) 38152 fordblks (free_size) 4252 uordblocks (used_size) 33900

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 2292, room 16 
tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4 
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   DOUT
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 0.
pp_task_hdl : 3ffeffe0, prio:14, stack:512
pm_task_hdl : 3ffef858, prio:1, stack:176
frc2_timer_task_hdl:0x3fff5108, prio:12, stack:200

RavenSystem, 2020-09-21:
Fixed with v3.3.2.

Many thanks for feedback and logs!

closed 2020-09-21 by RavenSystem

sorriso93, 2020-09-21:
Thanks to you for you work