RavenSystem/esp-homekit-devices #1146: Configuration page Fixed with v3.3.2 does not save wifi configuration

sorriso93, 2020-09-21:
Configuration page Fixed with v3.3.2 does not save wifi configuration, goes always back to setup mode...

Home Accessory Architect v3.3.2
Developed by José Antonio Jiménez Campos (@RavenSystem)

SETUP MODE
WiFi config init
No WiFi config found
Start AP SSID=HAA-1D83B4
Start DHCP server
mode : sta(dc:4f:22:1d:83:b4) + softAP(de:4f:22:1d:83:b4)
add if0
add if1
bcn 100
Start DNS server
Start HTTP server
Start WiFi scan
scandone

Many thanks for feedback and logs!

_Originally posted by @RavenSystem in https://github.com/RavenSystem/esp-homekit-devices/issues/1145#issuecomment-695987006_

RavenSystem, 2020-09-21:
Maybe 3.3.1 broke sysparam storage. Enable Reset Settings, click Save, and enter setup again to configure Wifi.

RavenSystem, 2020-09-21:
Nevermind. Update to v3.3.3.

closed 2020-09-21 by RavenSystem