RavenSystem/esp-homekit-devices #1147: Support for Adaptive Lighting?

macjl, 2020-09-22:
Hello,

I understood that Adaptive Lighting, that was added to iOS 14, would need an update on accessories side to be enabled. 
Do you know if you will be able to provide such support for lightbulb devices with HAA?

bj-tracer, 2020-09-23:
Hmmm...
It's a homekit automation function as i know.
Device must have color temperature adjustment only.

macjl, 2020-09-23:
Not only. There are new characteristics for that : 
https://github.com/ebaauw/homebridge-hue/issues/766

lcges, 2020-10-10:
Hey, are you planning to add Adaptive Lighting support to your driver?

MikeyBally, 2020-10-28:
Would love this!

macjl, 2020-11-19:
FYI, homebridge-hue is now working with Adaptive Lightning : https://github.com/ebaauw/homebridge-hue/issues/766
I juste tested it, and it works nice! Maybe you could find informations about how to include it in HAA in the repository?

mwwgithub, 2020-12-30:
This would be a game changer for me.  Right now I’m primarily using WiZ bulbs and recessed lights that are exposed to HomeKit via HomeBridge, partly because I like that the WiZ lights can autonomously follow a set rhythm for color temperature.  

However the Feit recessed lights that can run this firmware have a much nicer design with a recessed lens to reduce glare.  But I’d miss the “follow the sun” color temperature.

lcges, 2021-01-21:
What status is the thread on?

mwwgithub, 2021-04-24:
I'd love to know the same.  Any chance this can be implemented?

RavenSystem, 2021-06-04:
I have no plans to include Apple Adaptive Lighting. However, you can use timetable actions and lightbulb service notification to create and customize your own Adaptive Lighting system.

closed 2021-06-04 by RavenSystem