RavenSystem/esp-homekit-devices #1148: Shelly 1 as Stateless Button

Quoshak, 2020-09-23:
Hi,

I would like to setup my shelly 1 as a "stateless button" and have external push button connected as a trigger. With that I would like to setup different actions for single click, double click and long press.

Is that possible? If it is, can you help me with JSON?

Many thanks!

Quoshak, 2020-09-23:
I was trying to assemble correct JSON for this case, but it does not work. 
I can see a device in HomeApp, I can set actions for all three types, but neither tapping in HomeApp, neither pushing button works.

`{"a":[{"t":3,"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"f0":[{"g":5,"t":0}],"f1":[{"g":5,"t":1}]}]}`

I guess I don't understand docs and device enough. Please help.

RavenSystem, 2020-09-23:
`{"a":[{"t":3,"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"f0":[{"g":5,"p":0,"i":1,"t":1}],"f1":[{"g":5,"t":2}],"f2":[{"g":5,"t":3}]}]}`

Quoshak, 2020-09-23:
Works like charm! Thank you @RavenSystem!

Of course there is always follow up... :)

Can one action do two operation depending on current state? 
e.g. if first single click turn light on, can next single click turn light off?

freddair, 2020-09-24:
That can be done by Homekit automations either via home app (convert to shortcut and use if-then) or eve app (using conditions).

Quoshak, 2020-09-24:
Works! Thank you.

closed 2020-09-24 by Quoshak