RavenSystem/esp-homekit-devices #1150: Problem with Light Bulb

lougovsk, 2020-09-25:
Hello folks, 

I've converted 2 Tuya bulbs (i believe these https://templates.blakadder.com/lohas-ZN012.html) using Tuya-convert. After initial configuration they both appeared on my network. As json config i used {"c":{"b":[{"g":0}]},"a":[{"t":30,"r":14,"g":12,"v":13,"w":4,"fr":4,"fg":4,"fv":4}]}. As a result I have two bulbs in slightly different states. One I was able to successfully add to the HomeKit, but I cannot turn it on or off. Second one I cannot find through the HK (even though they are both connected to my router). Trying to access configuration page doesn't work for neither of them. I have used nmap to scan ports on the first one and the only open tcp port is 5556. Turning them on and off 8 times doesn't lead them to the Setup Mode (at least I don't see corresponding WiFi networks). Running nc -kulnw0 45678 doesn't give any output, so I believe there is no flashing going on. I've also checked my router traffic monitor and according to it both devices downloaded around 1.7 MB yesterday, but after that they were not attempting to access internet anymore. 
Any clues or suggestions?

wayn3, 2020-09-27:
The bulb you have linked appears to be using digital led controls via the [MY92xx family](http://www.my-semi.com/content/products/product_list.aspx?id=2), which isn't supported by HAA but support is planned for the WS2812b on the [TODO-list](https://github.com/RavenSystem/esp-homekit-devices/wiki/TODO-List) so who knows what the future may bring...

lougovsk, 2020-09-28:
Hello wayn3, thanks for the info. It kinda explains my problems.

lougovsk, 2020-09-29:
@RavenSystem, I have a problem with debugging. Kinda not getting any output over UART. Interestingly, that before I do initial configuration i at least get some messages (like scan done and so on). But after configuring the device nothing comes anymore. Any clues? do I need to compile with special flags? Can it be related to a misconfiguration (that would be a bit strange cause device is up and running or at least I can add it to the home app)?

lougovsk, 2020-10-01:
Never mind. Have figured it out. In JSON  config string „o“ option was missing in general config section .

closed 2020-10-14 by lougovsk