RavenSystem/esp-homekit-devices #1152: Sonoff basic shows up in HomeKit but pairing fails

FelixSeidel, 2020-10-01:
Please help,
I have tried everything but I can't get my Sonoff Basic to pair with HomeKit.

I have successfully connected it to the network. I also see it as a switch in HomeKit under "Add Device" and can access it via IP when I activate Setup Mode.

When I try to add Sonoff as a switch, the loader rotates for about a minute, then it shows "Device Not reachable" even though it was previously suggested.
When I scan the QR the same message appears, but faster.

I have even tried to format the JSON correctly or as a long string. No difference.
I tried it with an iOS 14.0.1 device and an iOS 13.6.1 device. Again no difference.

My JSON looks like this:
`{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}],"s":1}]}`

wayn3, 2020-10-01:
json looks fine, attempt to bring the device closer to your wifi ap just in case signal is an issue, other than that scan the code _not the QR_

try rebooting the device, your router and phone, make sure device is using the latest firmware however no obvious issue here

quartist, 2020-10-03:
Hi, 

your iPhone is in the same network with sonoff? The same SSID? In my opinion this may be a problem with mDNS or you have enabled on router "client isolation". Wireless Client Isolation is a security feature that prevents wireless clients from communicating with one another.

closed 2020-11-03 by stale[bot]