RavenSystem/esp-homekit-devices #1154: Esooli light switch analogous to sonoff T1

ArchiMsk, 2020-10-02:
[image]
After flashing, the switch often becomes unavailable in homekit, brand Esooli switch, firmware 3.4.6. Other sonoff switches work in homekit without problems.
[image]

nickforev, 2020-10-15:
Press and hold the sensor for about 15-17 seconds (if there is a buzzer on the board, it will beep once, and after 5 seconds a double beep).
After that, find your switch for mac and ip, and go to the setup page, tick the reset wifi and reset homekit id.
save.
Then find a new HAA network and set up as new ...

closed 2020-11-14 by stale[bot]

mixza2, 2021-11-01:
Can someone throw me json for this switch ?(1gang)