RavenSystem/esp-homekit-devices #1158: Shelly RGBW2 - tested JSON code?

kompak23, 2020-10-05:
Hi, 
I already have one Shelly RGBW2 in my home, but with old software (1.x), and it works OK, but I want next one and there is a problem. I flashed it properly, and device is working - it is connected to my WIFI and setup mode is still active. I've pasted JSON code: {"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}],"s":0}]} from https://glumb.github.io/haa-configurator/
Due to the active setup mode, I can't add accessory to Homekit. 
Can you help me? 
What can couse this issue?

kx3kx3kx3, 2020-10-05:
this is the right json for shelly rgbw {"c":{"l":2,"b":[{"g":13,"t":5}]},"a":[{"t":30,"r":15,"g":12,"v":14,"w":4,"b":[{"g":5,"p":0},{"g":5,"t":0},{"g":13}]}]}

closed 2020-11-05 by stale[bot]

mrmaximas, 2021-03-22:
> this is the right json for shelly rgbw {"c":{"l":2,"b":[{"g":13,"t":5}]},"a":[{"t":30,"r":15,"g":12,"v":14,"w":4,"b":[{"g":5,"p":0},{"g":5,"t":0},{"g":13}]}]}

Please help me to make a proper json for rgbw2 with white channel only. Thanks!