RavenSystem/esp-homekit-devices #1162: Lock accessory opens after reboot

Kienz, 2020-10-10:
I've updated from 1.6.6 to 3.5.7 with the identical configuration.
```json
{"a":[{"t":4,"i":4,"0":{"r":[{"g":4,"v":1,"i":1.5}]}},{"t":6,"i":10,"f1":[{"g":5,"p":0,"t":0}]},{"1":{"s":[{"a":1}]}},{"1":{"s":[{"a":2}]}}]}
```
After the update the lock opens and never closes. To close the mechanic lock I have to press the lock accessory in HomeKit. What can I do to prevent opening the lock on reboot/startup?
I tried adding "xa":0 to lock accessory - no success.

RavenSystem, 2020-10-10:
`{"c":{"0":{"r"[{"g":4}]}},"a":[{"t":4,"i":4,"0":{"r":[{"g":4,"v":1,"i":1.5}]}},{"t":6,"i":10,"f1":[{"g":5,"p":0,"t":0}]},{"1":{"s":[{"a":1}]}},{"1":{"s":[{"a":2}]}}]}`

Kienz, 2020-10-10:
You missed ":" after "r".
```json
{"c":{"0":{"r":[{"g":4}]}},"a":[{"t":4,"i":4,"0":{"r":[{"g":4,"v":1,"i":1.5}]}},{"t":6,"i":10,"f1":[{"g":5,"p":0,"t":0}]},{"1":{"s":[{"a":1}]}},{"1":{"s":[{"a":2}]}}]}
```

Thank you very much for the fast response. It works!

closed 2020-10-10 by Kienz