RavenSystem/esp-homekit-devices #1164: Release HAA_3.5.7 - wifi settings can't save

komarofleo, 2020-10-10:
Hi all. Help needed! After flash the Release HAA_3.5.7 - wifi settings can't save. No WiFi config found and starting AP. 
When I add ssd and password I will hit SAVE - and nothing happend. Help!
I was trying it any phone, and iPad - still the same.

RavenSystem, 2020-10-12:
Try last version.

komarofleo, 2020-10-12:
Dear Jose, in old version after successful flashed chip I can reach in setup mode to make some corrections in json.
I print 192.168.4.1 into browser and could reach web-interface. 
Now I need to browse 192.168.31.117 (etc) ip address to get web-interface instead of http://haa-number.local and  192.168.4.1
Is it possible to reach web-interface via local network throw 192.168.4.1 or http://haa-number.local?
Or is it possible to use old versions?
Thank you so much

komarofleo, 2020-10-12:
So after flashing - I can't change WiFi settings via web-interface - the SAVE green button doesn't respond.

Murdrock, 2020-10-13:
> So after flashing - I can't change WiFi settings via web-interface - the SAVE green button doesn't respond.

I believe there was a bug in 3.5.7 about saving wifi settings. Try flashing the 3.5.8 the wifi save setting is fixed.

komarofleo, 2020-10-13:
I have flashed with 3.5.8 - the same problem. I cant change wifi settings second time after flashing. - the Save button does not respond.

tonysprenk, 2020-10-13:
Post your json. Could be a syntax or config error.

mariusvasss, 2020-10-13:
Erase flash then write the firmware.

komarofleo, 2020-10-13:
I've done erasiing firmware 5 times - result the same! On several chips. 
Json the same and works fine on previous versions - here:
{
  "c": { "l": 0, "b": [{ "g": 2, "p": 0, "t": 5 }] },
  "a": [{
    "t": 45,
    "o": 1,
    "c": 1,
    "f": 10,
    "0": { "r": [{ "g": 15, "v": 1 }, { "g": 4 }] },
    "1": { "r": [{ "g": 15 }, { "g": 4, "v": 1 }] },
    "2": { "r": [{ "g": 15 }, { "g": 4 }] },
    "3": { "r": [{ "g": 15, "i": 1 }, { "g": 4 }] },
    "4": { "r": [{ "g": 15 }, { "g": 4, "i": 1 }] },
    "f0": [{ "g": 5, "p": 0 }],
    "f1": [{ "g": 13, "p": 0 }],
    "f2": [{ "g": 5, "t": 0 }, { "g": 13, "t": 0 }]
  }]
}

komarofleo, 2020-10-13:
The first time when I am flashin and input wifi - is okay - the chip starts downloading code and after 6 min all is okay and I can add it to Apple Home. But I neeed another wifie network and I can't do it via setup mode. Help pls!

komarofleo, 2020-10-13:
I think I found the solution! When wifi is ok - I can't reach 192.168.4.1 but can see settings via local network ip:192.168.31.117
So if you wanna see settings on 192.168.4.1 - you need turn of the router )))

RavenSystem, 2020-10-13:
That is explained in the Wiki, and it is not a bug: https://github.com/RavenSystem/esp-homekit-devices/wiki/Setup-Mode#device-connection

closed 2020-10-13 by RavenSystem

exrector, 2020-11-07:
no. This is mistake. I have 3 devices. I have changed the router and SSID and I cannot configure any device to the new network