RavenSystem/esp-homekit-devices #1166: Shelly Dimmer crashes / freezes

freddair, 2020-10-11:
hi, I think since 3.5.7 and now 3.5.8 my shelly dimmer1 occasionally does not respond to switch operations and comes back after a few seconds. Sometimes it also won't switch the light off but loops on/off/on/off... instead. any idea? Here is a log output, maybe it helps:
```
Advanced ESP Logger (c) 2020 José Antonio Jiménez Campos

0003.563 10.0.1.45-D45DAC: Free Heap: 43500
0003.565 10.0.1.45-D45DAC: Start HK Server
0003.567 10.0.1.45-D45DAC: Free Heap: 34332
0003.568 10.0.1.45-D45DAC: Exec Action: acc 0, action 2
0003.569 10.0.1.45-D45DAC: Starting server
0004.074 10.0.1.45-D45DAC: mDNS reannounced
0004.157 10.0.1.45-D45DAC: mDNS announcement: Name=HAA-D45DAC md=RavenSystem HAApv=1.1id=17:B9:2D:DB:57:ECc#=2s#=1ff=0sf=0ci=5
                                              sh=ZnaKgQ== Port=5556 TTL=240
0004.347 10.0.1.45-D45DAC: [5] New 10.0.1.34:62692
0004.350 10.0.1.45-D45DAC: [5] Verify 1/2
0004.689 10.0.1.45-D45DAC: [6] New 10.0.1.29:5824
0004.693 10.0.1.45-D45DAC: [6] Verify 1/2
0005.028 10.0.1.45-D45DAC: [5] Verify 2/2
0005.032 10.0.1.45-D45DAC: [5] Found pairing with FB97E758-4A61-4E1A-9117-69F28BAAE055
0005.207 10.0.1.45-D45DAC: [5] Verification OK
0005.210 10.0.1.45-D45DAC: [6] Verify 2/2
0005.336 10.0.1.45-D45DAC: Saving states
0005.399 10.0.1.45-D45DAC: [6] Verification OK
0005.403 10.0.1.45-D45DAC: [5] Get Accessories
0005.437 10.0.1.45-D45DAC: [6] Get Accessories
0005.522 10.0.1.45-D45DAC: [5] Update Characteristics
0005.558 10.0.1.45-D45DAC: [5] Update Characteristics
0005.608 10.0.1.45-D45DAC: [6] Update Characteristics
0005.667 10.0.1.45-D45DAC: [6] Update Characteristics
0008.549 10.0.1.45-D45DAC: [5] Get Characteristics
0019.190 10.0.1.45-D45DAC: [7] New 10.0.1.26:18369
0019.193 10.0.1.45-D45DAC: [7] Verify 1/2
0019.535 10.0.1.45-D45DAC: [8] New 10.0.1.31:33497
0019.539 10.0.1.45-D45DAC: [8] Verify 1/2
0019.871 10.0.1.45-D45DAC: [7] Verify 2/2
0019.875 10.0.1.45-D45DAC: [7] Found pairing with 6AA2AD33-9318-4C98-920A-C52A60B90135
0020.052 10.0.1.45-D45DAC: [7] Verification OK
0020.059 10.0.1.45-D45DAC: [8] Found pairing with FB97E758-4A61-4E1A-9117-69F28BAAE055
0020.233 10.0.1.45-D45DAC: [8] Verification OK
0020.237 10.0.1.45-D45DAC: [7] Get Accessories
0020.262 10.0.1.45-D45DAC: [8] Get Accessories
0020.921 10.0.1.45-D45DAC: [8] Get Characteristics
0020.982 10.0.1.45-D45DAC: [8] Get Characteristics
0026.301 10.0.1.45-D45DAC: [5] Get Characteristics
0058.385 10.0.1.45-D45DAC: DigI GPIO 12
0058.387 10.0.1.45-D45DAC: Target RGBW = 0, 0, 0, 65534, 0, 0
0058.388 10.0.1.45-D45DAC: Exec Action: acc 1, action 1
0058.390 10.0.1.45-D45DAC: Wildcard 0 100.00
0058.391 10.0.1.45-D45DAC: Wilcard Action 0 100.00
0063.385 10.0.1.45-D45DAC: Saving states
0066.745 10.0.1.45-D45DAC: DigI GPIO 12
0067.245 10.0.1.45-D45DAC: Target RGBW = 0, 0, 0, 0, 0, 0
0067.249 10.0.1.45-D45DAC: UART Action
0072.245 10.0.1.45-D45DAC: Saving states
0088.576 10.0.1.45-D45DAC: [5] Update Characteristics
0088.609 10.0.1.45-D45DAC: [5] Update Characteristics
0089.289 10.0.1.45-D45DAC: [6] Update Characteristics
0089.379 10.0.1.45-D45DAC: [6] Update Characteristics
0089.503 10.0.1.45-D45DAC: DigI GPIO 12
0089.506 10.0.1.45-D45DAC: Target RGBW = 0, 0, 0, 65534, 0, 0
0089.507 10.0.1.45-D45DAC: Exec Action: acc 1, action 1
0089.509 10.0.1.45-D45DAC: Wildcard 0 100.00
0089.511 10.0.1.45-D45DAC: Wilcard Action 0 100.00
0089.512 10.0.1.45-D45DAC: Exec Action: acc 1, action 52
0089.514 10.0.1.45-D45DAC: UART Action
0094.503 10.0.1.45-D45DAC: Saving states
0124.193 10.0.1.45-D45DAC: mDNS reannounced
0124.749 10.0.1.45-D45DAC: [7] Update Characteristics
0124.900 10.0.1.45-D45DAC: [7] Update Characteristics
0125.067 10.0.1.45-D45DAC: [7] Get Characteristics
0125.123 10.0.1.45-D45DAC: [7] Get Characteristics
[... 176 lines cut ...]
0855.214 10.0.1.45-D45DAC: Wilcard Action 0 100.00
0855.588 10.0.1.45-D45DAC: DigI GPIO 12
0856.088 10.0.1.45-D45DAC: Target RGBW = 0, 0, 0, 0, 0, 0
0856.090 10.0.1.45-D45DAC: Exec Action: acc 1, action 0
0856.091 10.0.1.45-D45DAC: UART Action
0856.648 10.0.1.45-D45DAC: DigI GPIO 12
0856.650 10.0.1.45-D45DAC: Target RGBW = 0, 0, 0, 65534, 0, 0
0856.651 10.0.1.45-D45DAC: Exec Action: acc 1, action 1
0856.653 10.0.1.45-D45DAC: Wildcard 0 100.00
0856.654 10.0.1.45-D45DAC: Wilcard Action 0 100.00
0857.208 10.0.1.45-D45DAC: DigI GPIO 12
0857.697 10.0.1.45-D45DAC: Setup mode call
0857.702 10.0.1.45-D45DAC: 
0857.705 10.0.1.45-D45DAC: Rebooting...
0857.706 10.0.1.45-D45DAC: 
0857.708 10.0.1.45-D45DAC: Target RGBW = 0, 0, 0, 0, 0, 0
0858.308 10.0.1.45-D45DAC: DigI GPIO 12
0858.310 10.0.1.45-D45DAC: Target RGBW = 0, 0, 0, 65534, 0, 0
0858.311 10.0.1.45-D45DAC: Exec Action: acc 1, action 1
0858.313 10.0.1.45-D45DAC: Wildcard 0 100.00
0858.314 10.0.1.45-D45DAC: Wilcard Action 0 100.00
0859.967 10.0.1.45-D45DAC: DigI GPIO 12
0860.468 10.0.1.45-D45DAC: Target RGBW = 0, 0, 0, 0, 0, 0
0860.471 10.0.1.45-D45DAC: UART Action
```

RavenSystem, 2020-10-11:
Try adding a harder input filter (`"f":30`), because Shelly inputs are very sensible.

freddair, 2020-10-13:
Not better :-(

miamilabs, 2020-10-21:
Is it possible that you share your config?
I expect same issue with my dimmer.

It only turns on/off in loop than stops.

Can i use the screen function while the 230v are attached?

freddair, 2020-10-22:
Sure, here is my config. I only use input 1 as on/off switch. Maybe anyone finds a clue what might be wrong?

{"c":{"o":0,"n":"ShellyDimmer-Flur","f":30,"r":[{"n":0}],"l":15,"i":0,"b":[{"g":13,"t":5}]},"a":[{"t":30,"d":0.1,"b":[{"g":12,"i":1,"p":0},{"g":12,"t":0}],"0":{"u":[{"v":"010202060000f401000000ff04"}]},"y0":[{"v":1,"0":{"u":[{"v":"010202060a00f4010000010904"}]}},{"v":5,"0":{"u":[{"v":"010202063200f4010000013104"}]}},{"v":10,"0":{"u":[{"v":"010202066400f4010000016304"}]}},{"v":15,"0":{"u":[{"v":"010202069600f4010000019504"}]}},{"v":20,"0":{"u":[{"v":"01020206c800f401000001c704"}]}},{"v":25,"0":{"u":[{"v":"01020206fa00f401000001f904"}]}},{"v":30,"0":{"u":[{"v":"010202062c01f4010000012c04"}]}},{"v":35,"0":{"u":[{"v":"010202065e01f4010000015e04"}]}},{"v":40,"0":{"u":[{"v":"010202069001f4010000019004"}]}},{"v":45,"0":{"u":[{"v":"01020206c201f401000001c204"}]}},{"v":50,"0":{"u":[{"v":"01020206f401f401000001f404"}]}},{"v":55,"0":{"u":[{"v":"010202062602f4010000012704"}]}},{"v":60,"0":{"u":[{"v":"010202065802f4010000018b04"}]}},{"v":65,"0":{"u":[{"v":"010202068a02f4010000010904"}]}},{"v":70,"0":{"u":[{"v":"01020206bc02f401000001bd04"}]}},{"v":75,"0":{"u":[{"v":"01020206ee02f401000001ef04"}]}},{"v":80,"0":{"u":[{"v":"010202060032f4010000013104"}]}},{"v":85,"0":{"u":[{"v":"010202065203f4010000015404"}]}},{"v":90,"0":{"u":[{"v":"010202068403f4010000018604"}]}},{"v":95,"0":{"u":[{"v":"01020206b603f401000001b804"}]}},{"v":100,"0":{"u":[{"v":"01020206e803f401000001ea04"}]}}]} ,{"i":0.5,"1":{"s":[{"a":1}]}}]}

freddair, 2020-10-24:
I'm about to flash the Shelly Dimmer back to stock as it is too buggy on HAA - sadly...

miamilabs, 2020-10-24:
> Sure, here is my config. I only use input 1 as on/off switch. Maybe anyone finds a clue what might be wrong? [...]

If fixed my dimmer problem.
Problem 1) I fix it with reflash of whole dimmer, my contact of 1 pin was not that good so the flash was broken.
Problem 2 [flashing light] ) My Bulb was LED with 8 watt. This seems to be low for the dimmer, after i replaced the bulb with regular one, it fixed my problem.

closed 2020-11-09 by RavenSystem