RavenSystem/esp-homekit-devices #1168: Change Wi-Fi settings, how?

komarofleo, 2020-10-13:
Anybody pls, help. How can I change the wi-fi settings after flashing?!
My steps:
1. Input via browser http://HAA-number.local
2. Remove wifi settings
3. Entering setup mode - AP + 192.168.4.1 
4. Input new ssd and wifi password and hit save and... NOTHING (chip not rebooting)
So I cant change wi-fi settings. 

screenshots on usb to 
[image]
[image]
[image]
[image]
[image]

monitor and phone screen are attached here...

rjblake, 2020-10-15:
I've found in the past if the JSON is not correct it would not save. Share the JSON?

komarofleo, 2020-10-15:
JSON is correct and perfectly works 
{
  "c": { "l": 0, "b": [{ "g": 2, "p": 0, "t": 5 }] },
  "a": [{
    "t": 45,
    "o": 1,
    "c": 1,
    "f": 10,
    "0": { "r": [{ "g": 15, "v": 1 }, { "g": 4 }] },
    "1": { "r": [{ "g": 15 }, { "g": 4, "v": 1 }] },
    "2": { "r": [{ "g": 15 }, { "g": 4 }] },
    "3": { "r": [{ "g": 15, "i": 1 }, { "g": 4 }] },
    "4": { "r": [{ "g": 15 }, { "g": 4, "i": 1 }] },
    "f0": [{ "g": 5, "p": 0 }],
    "f1": [{ "g": 13, "p": 0 }],
    "f2": [{ "g": 5, "t": 0 }, { "g": 13, "t": 0 }]
  }]
}

komarofleo, 2020-10-15:
The problem is absent in previous version, So I cant change wifi (

rjblake, 2020-10-15:
Odd - I've just changed the WiFi to another SSID on my network. It saved without issue, ESP8266 restarted and connected to new SSID. Went back in and changed again and working correctly. I did this through Chrome browser on Mac, did not tick the box "Remove WiFi Settings", just selected the new SSID and entered password

komarofleo, 2020-10-15:
Can you tell me - how you have entered to setup mode?
And how did you connected to web interface of esp (haa-xxx or 192.168.4.1)

rjblake, 2020-10-15:
To enter setup mode, I have a button configured in HomeKit, but I also tried using Emergency (press Reset button on ESP8266 twice within 3 seconds). For the initial connection I accessed using 192.168.4.1 and after that the IP address assigned from my DHCP server

From a new install on the ESP8266:
1. Restart the ESP8266, on computer connect WiFi to device (192.168.4.1)
2. Configure JSON as needed
3. Click on "WiFi Networks" button to refresh SSIDs available
4. Select SSID and enter password for network
5. Click on "Save"
6. Device restarts and connects to network
7. To change network connection from this stage, restart device (press reset 2x on ESP8266 in less than 3 secs)
8. Connect using browser to assigned IP and change as needed. If at different location, I'd start from step 1

komarofleo, 2020-10-15:
Thank u, I've done it 100 times before and all was vonderfull. But now - I can't pass the 5-th step - when I press SAVE (green button) - it is stack and nothing happens, data not saved. I was trying 5 different modules, sad....

rjblake, 2020-10-15:
Did you try going back to earlier version?

komarofleo, 2020-10-15:
How can I do it? Is it possible to downgrade? Please teel me!

rjblake, 2020-10-15:
You can download an earlier version you know worked from this link [here](https://github.com/RavenSystem/haa/releases). The go through the initial install using 
`esptool.py -p /dev/<your_ESPPort> erase_flash`
`esptool.py -p /dev/<your_ESPPort> --baud 115200 write_flash -fs 1MB -fm dout -ff 40m 0x0 fullhaaboot.bin`
with the version of `fullhaaboot.bin` you downloaded. Then follow instructions under "Installation without OTA"
Make sure to untick Auto OTA updates (just in case). Alternatively add all files to a "Custom Server" so I can easily change to a different version if needed

komarofleo, 2020-10-15:
Thanx a lot man - I will try!

rjblake, 2020-10-15:
Also use compact JSON to save white space and memory:
`{"c":{"l":0,"b":[{"g":2,"p":0,"t":5}]},"a":[{"t":45,"o":1,"c":1,"f":10,"0":{"r":[{"g":15,"v":1},{"g":4}]},"1":{"r":[{"g":15},{"g":4,"v":1}]},"2":{"r":[{"g":15},{"g":4}]},"3":{"r":[{"g":15,"i":1},{"g":4}]},"4":{"r":[{"g":15},{"g":4,"i":1}]},"f0":[{"g":5,"p":0}],"f1":[{"g":13,"p":0}],"f2":[{"g":5,"t":0},{"g":13,"t":0}]}]}`

Partizan7676, 2020-10-15:
@komarofleo try to make settings as far as possible from wifi - it helps in my case. dont know why, but better use 3.5.8 - is more stable. a open an issue for that. todays 3.5.10 also have that bug - device no response, only if access it on its AP(to which device is connected)

komarofleo, 2020-10-15:
So I can definitely say - When I hit Reset WiFi - I can't then press Save button and update wifi settings - it does/nt respond and dont want to reboot with new wi-f- config.

komarofleo, 2020-10-15:
So - thanx all of you. I finally found the solution. It's work perfect for me.
1. I have flashed with old version 3-3-3 wihout OTA
2. All works perfect - wifi settings are stored easy!!!

Thank you all - wish for somebody it will be useful.

komarofleo, 2020-10-15:
rjblake - thank you man!!! Can you text me via social network (whatsup or viber pls. or mail me - 80509048505@mail.ru)

Partizan7676, 2020-10-16:
try 3.5.8 and get same issue - settings not saved. flashing 3.3.3 - settings are apply, but device dont have remote access. i.e. i switch my iphone to LTE network and get no response from 3.3.3. what is strange it is connected to the same AP as device with firmware 3.5.8 and one device response, other not. any idea?

Partizan7676, 2020-10-16:
got it!
This must be included in wiki as preparation stage. in my case domain was not "local" it was ubnt and i add ubnt.local so that is out of rules for mDNS. thats all ))) check it simple. for ex you add device HAA-xxxx, just try to ping it by its name with .local ie HAA-xxxx.local when it says unable to resolve name - it means that your local domain is different and multicast to 224.0.0.251 port 5353 UDP are blocked.
@RavenSystem you fully described router config, but that moment will be great to add to wiki too

rjblake, 2020-10-16:
> try 3.5.8 and get same issue - settings not saved. flashing 3.3.3 - settings are apply, but device dont have remote access. i.e. i switch my iphone to LTE network and get no response from 3.3.3. [...]

@Partizan7676 - since you are using Ubiquiti equipment (not sure what firmware versions or controller version), you may want to check the following (had issues with my IoT devices):

Under Settings->Wireless Networks->Advanced, look for PMF and make sure it is Disabled (unless of course you need it for other reasons). In my case this solved some problems. I have 2x test devices running on 3.5.10 and have no issues with "no response" or changing and saving changes to WiFi

peros550, 2020-10-18:
guys I have the same issue. I am unable to flash by wire. Did you find any solution?

Partizan7676, 2020-10-18:
@peros550 check mDNS in your network

peros550, 2020-10-18:
@Partizan7676 I have a device which I tried to configure on an another SSID. I tried reset wifi settings, and now the device always startup on setup mode and announce its AP. When I connect to it , it searches for available wifi settings, I try to select a new on but the device does not save my settings, hence it remains in AP mode. 

Does all this make sense to you? mDNS would be an option if the device was connected to my router, but that's not the case :(

komarofleo, 2020-10-18:
New release where launched just now - be happy )))

ali199011, 2020-10-21:
Wi‑Fi

ali199011, 2020-10-21:
sky.net/status

ali199011, 2020-10-21:
sky.net/status

closed 2020-11-05 by RavenSystem