RavenSystem/esp-homekit-devices #1170: No Reponse issues since HAA 3.5.6

huta76, 2020-10-15:
I have noticed that since 3.5.6, I am getting peristent No reponse issues.  If I reboot router, all comes back to life but within an hour both switches turn to No response.  My switches are Sonoff Basic R2's if that helps.  Anyone else has similar issues?

Partizan7676, 2020-10-15:
same here/ but i find that version 3.5.8 - works fine. 3.5.9 - not. will test how 3.5.10 works...

Partizan7676, 2020-10-15:
3.5.10 have same bug

Partizan7676, 2020-10-16:
just check last message here - https://github.com/RavenSystem/esp-homekit-devices/issues/1168

huta76, 2020-10-16:
@Partizan7676 can you elaborate wha you did please? Did you set your domain in dhcp as local?

Partizan7676, 2020-10-18:
@huta76 just check how mDNS work in your lan. Just ping your device with its hostname ex: HAA-xxxx if you get result "couldn't resolve hostname" that it! check your router config for mDNS settings and check ruleset for blocking udp multicast with address 224.0.0.251 and port 5353

huta76, 2020-10-22:
Thanks again @Partizan7676 .  Seems like my issue is adding my 2 LG TV's to the Home.   Without them added, all works fine.   Once they're added and powered off, the No Response issues start again.  I'm suspecting some bug in the TV firmware.

closed 2020-11-21 by stale[bot]

huta76, 2020-11-27:
No response issues persisted.  It turned out that my cable modem (Arris) was at fault.  Got a TPlink Archer A10, and everything has been running smoothly since then.   If your router does not handle MDNS well, changing it is the only option.