RavenSystem/esp-homekit-devices #1171: Sonoff sv contact sensor stops working

karnas99, 2020-10-15:
Hi.

I build two contact sensor on gpio 5 and gpio 4. The issue is that it is randomly stopping working. After restart it is usually working but after some times it stops.

config:
{"c":{"l":13,"i":0,"b":[{"g":0,"t":1}]},"a":[{"t":5,"f1":[{"g":5,"t":0}],"f0":[{"g":5}]},{"t":5,"f1":[{"g":4,"t":0}],"f0":[{"g":4}]}]}

Any ideas why it happens?

RavenSystem, 2020-11-09:
Update to last version.

closed 2020-11-09 by RavenSystem

karnas99, 2020-11-25:
Thanks for advise I updated to last ver. However it is till not perfect. Sometimes contact sensor does not react and it starts to react after some time. I mean I close contact sensor and nothing happen. Theree minutes later I try it again and it is working. It seems to me like if sonoff is somehow sleeping and then it takes some time he wakes up and it is working. 

Is there any additional configuration which keeps sonoff awaken and responsive?