RavenSystem/esp-homekit-devices #1173: Garage Door with Shelly 1 and Push button

aleroot, 2020-10-15:
Hi,I have a garage door I would want to automate with homekit, I have connected the garage door two wires to the I and O pin on the Shelly 1 and the push button on the SW, the problem is that I cannot make it work correctly since the two wires coming from the Garage Door are supposed to be closed(short circuit them) for a short period of time each time you want to send the command to open or close the garage door, I have tried with this configuration below: 

`{
  "c": {
    "b": [
      {
        "g": 5,
        "p": 0,
        "i": 1,
        "t": 5
      }
    ]
  },
  "a": [
    {
      "0": {
        "r": [
          {
            "g": 4,
            "v": 0
          }
        ]
      },
      "1": {
        "r": [
          {
            "g": 4,
            "v": 1,
            "i":0.5
          }
        ]
      },
      "2": {
        "r": [
          {
            "g": 4,
            "i": 1.5
          },
          {
            "g": 4,
            "v": 1,
            "i": 0.5
          },
          {
            "g": 4,
            "v": 1,
            "i": 2
          }
        ]
      },
      "3": {
        "a": 2
      },
      "s": 5,
      "t": 40,
      "d": 20,
      "b": [
        {
          "g": 5
        }
      ]
    }
  ]
}`

but the problem is that when I open the garage door either from homekit Home app or from push button then the two contact I and O are remaining closed(short circuited) so I cannot close the door anymore ....

Is there a way to close the contacts just shortly when the Open or close command is coming and then leave the two contacts open( NOT touching) for all the time? Is there a timeout property to bring back the contact to open after a period of time, lets say a second ?

aleroot, 2020-10-15:
I have tried even in this way: 

`{
  "c": {
    "b": [
      {
        "g": 5,
        "p": 0,
        "i": 1,
        "t": 5
      }
    ]
  },
  "a": [
    {
      "0": {
        "r": [
          {
            "g": 4,
            "v": 1,
            "i":0.5
          }
        ]
      },
      "1": {
        "r": [
          {
            "g": 4,
            "v": 1,
            "i":0.5
          }
        ]
      },
      "2": {
        "r": [
          {
            "g": 4,
            "i": 1.5
          },
          {
            "g": 4,
            "v": 1,
            "i": 0.5
          },
          {
            "g": 4,
            "v": 1,
            "i": 2
          }
        ]
      },
      "3": {
        "a": 2
      },
      "s": 5,
      "t": 40,
      "d": 20,
      "b": [
        {
          "g": 5
        }
      ]
    }
  ]
}`

but the problem I have is that when shelly boot up the contact is by default closed and makes my Garage door to open, so in case of a mains blackout the garage door opens ... 

Is there a way to have v = 0 when the shelly boot up? How?

aleroot, 2020-10-16:
I have tried even with this configuration:

`{"c":{"b":[{"g":5,"p":0,"i":1,"t":5}]},"a":[{"0":{"r":[{"g":4,"v":1,"i":0.5}]},"1":{"r":[{"g":4,"v":1,"i":0.5}]},"2":{"r":[{"g":4,"i":1.5},{"g":4,"v":1,"i":0.5},{"g":4,"v":1,"i":2}]},"3":{"a":2},"4":{"g":4,"v":0},"5":{"a":4},"6":{"a":4},"7":{"a":4},"s":0,"t":40,"d":20,"b":[{"g":5}]}]}`

Same result, when the Shelly boots up: v = 1 on g = 4, Why ? 
Even if I set the s = 0 all is the same... 
There must be a way to boot up the shelly with  v = 0 on g = 4 when the accessory is 40 (Garage door). I don't really understand why it doesn't work except is a bug...

macjl, 2020-10-17:
Hello,
I have same kind of problem since some weeks. I think a bug was introduced in a recent firmware, because before all worked fine with the garage door and same Json on Shelly 1 :
`{"c":{"z":0,"o":3},"a":[{"t":40,"d":14,"f3":[{"g":5,"t":1,"p":0}],"f4":[{"g":5,"t":0,"p":0}],"0":{"r":[{"g":4,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":4,"i":1.5},{"g":4,"v":1,"i":0.5},{"g":4,"v":1,"i":2}]},"3":{"a":2}},{"t":1,"1":{"s":[{"a":1}]}}]}`

@RavenSystem, is there a known bug in last firmware?

aleroot, 2020-10-18:
@macjl do you know what version was still working ? We can try to compare the source code and see what could have broken it . Thanks for your valuable feedback ;)

macjl, 2020-10-18:
I'm not sure at all. But maybe 3.5.3 or 3.5.4.
Changelog may indicate changes in boot behaviour in theses releases, and it can correspond to the period that I first noticed the bug...

morphix9, 2020-10-18:
I'm also struggling with the Garage door for some time - it's buggy :(

aleroot, 2020-10-19:
The change that cause the issue seems to be indeed introduced with version 3.5.3, due to the change of the initialization of the accessory.

@macjl  Could you try with the following JSON and let me know if at boot the GPIO is now low and working correctly ? Thanks.

macjl, 2020-10-19:
Hello @aleroot,
I just tried the Json, but still have the issue. I'm keeping searching what's wrong.

ndv27, 2020-10-22:
Hello @aleroot,
I tested JSON below also and gate opens after boot:
`{"c":{"z":0,"o":3},"a":[{"t":40,"d":14,"4":{"r":[{"g":4}]},"f3":[{"g":5,"t":1,"p":0}],"f4":[{"g":5,"t":0,"p":0}],"0":{"r":[{"g":4,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":4,"i":1.5},{"g":4,"v":1,"i":0.5},{"g":4,"v":1,"i":2}]},"3":{"a":2}},{"t":1,"1":{"s":[{"a":1}]}}]}  `

aleroot, 2020-10-23:
I've raised a pull request that fixes the issue, attached to this ticket [here](https://drive.google.com/file/d/1Z_mirZnPu_uZUqQtJLQ9XEfQSlZ4IST6/view?usp=sharing) you can find the prebuilt binaries to try...

aleroot, 2020-10-23:
Pull request just been rejected, by @RavenSystem that maybe now can teach us how to make it work without the patch.

macjl, 2020-10-24:
I think I got it. I just add `"0":{"r":[{"g":4}]}` in the general config section, and seems to work. Here is my Json now  : 
`{"c":{"z":0,"o":3,"0":{"r":[{"g":4}]}},"a":[{"t":40,"d":14,"f3":[{"g":5,"t":1,"p":0}],"f4":[{"g":5,"t":0,"p":0}],"0":{"r":[{"g":4,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":4,"i":1.5},{"g":4,"v":1,"i":0.5},{"g":4,"v":1,"i":2}]},"3":{"a":2}},{"1":{"s":[{"a":1}]}}]}`

i3laze, 2020-10-25:
+1, we’re definitely suffering from v3.5.3 innovation:

Changelog:

#Now initial GPIO output state before accessories boot is last declared value of first action with a digital output array, starting with action "0" of config section. Before it was always LOW.

i3laze, 2020-10-25:
@RavenSystem , I see a logic here to preserve relay state after reboot. However you could declare that as Initial state or introduce as an option. 
Please fix/hardcode old behavior for garage type accessory, otherwise every garage door owner would have to reconfigure their devices.

RavenSystem, 2020-11-09:
Use action `"0"` in config section to set initial relay state.

closed 2020-11-09 by RavenSystem