RavenSystem/esp-homekit-devices #1174: bug: custom server for updates not working

Partizan7676, 2020-10-16:
Can somebody explain how it works?

1. make web server(192.168.88.4:8010) and put binaries(/haa_upd). guest access enabled
2. put device into setup mode
3. set up server for custom updates 192.168.88.4/haa_upd
4. set up port 8010
5. uncheck secure connection
Also i try set server url with port 192.168.88.4:8010/haa_upd. make checkboxes AutoOTA update, update all with OTA - acceess logs of my server is empty. with direct link i can download files from it. nothing works?

wayn3, 2020-10-16:
Any reason your wanting to use alternate port 8010 for the web server?

Can you access the web directory file listing from a browser if you goto http://192.168.88.4:8010/haa_upd ?

Partizan7676, 2020-10-18:
> Any reason your wanting to use alternate port 8010 for the web server?
> 
> Can you access the web directory file listing from a browser if you goto http://192.168.88.4:8010/haa_upd ?

80 and 8080 a busy with webui of QNAP. listing? no, but if you need to get file by name it works. client ask server for exact file name?

wayn3, 2020-10-18:
unsure exactly how it requests the file but by name would make sense, can't say I've tried alternate ports but don't think this should be an issue, so if you chuck a file address into a browser, http://192.168.88.4:8010/haa_upd/haaboot.bin it will download the file?

What version are you currently on?

When you say the logs are empty, do you mean nothing meaningful or literally nothing...? (if something please post what it is), if literally nothing then you might have firewall issues because you should be able to see at least something

Partizan7676, 2020-10-19:
@wayn3 thank for your attention. with link you tell - i can download files. version from 333 to 3510. testing from 3510. logs i mean for web server. he has records when i put haa files only.

wayn3, 2020-10-20:
How about HAA OTA logs from the device when you attempt an update? (https://github.com/RavenSystem/esp-homekit-devices/wiki/Installation#haa-ota-logs)

> When device is installing or updating, you can capture logs from HAA OTA directly from your network with this command under macOS and GNU/Linux:
> 
> nc -kulnw0 45678

closed 2020-11-19 by stale[bot]