RavenSystem/esp-homekit-devices #1176: Sonoff Basic TH16: does not start emergency mode

coyotanark, 2020-10-17:
Hi!

I repost as my original issue was wrongfully closed and can't be re-opened:
https://github.com/RavenSystem/esp-homekit-devices/issues/1108

I wonder why this has been closed and locked for reopening when the claim of emergency mode working in all versions is not backed by any data but my claim for an issue is actual and real, proving that emergency mode DOES NOT work in all versions:

RavenSystem commented 5 days ago
Emergency setup mode works fine with all versions.
 @RavenSystem RavenSystem closed this 5 days ago
Repository owner locked as resolved and limited conversation to collaborators 5 days ago

=> why was it restricted for reopening -_- ?

Please provide steps to collect the right data in order to get fixed and exit the bad state. 

#########################################################

Long story, short this will explain the other issue I open too:

unexpected router change, setup new wireless network with new SSID.
try to enable emergency mode on multiple Sonoff TH16: power on, quickly power off (within 3 seconds) then power on again
What is expected:

wifi to connect to as the sonoff should be in Access Point Mode
update wifi settings and the device will find back normal working state.
What happens:

unsure but no wifi network to connect to. This affects 3 different devices, setup with 3 slightly different configurations and 3 different firmware versions (I see in the home app from when they were still there)
Kitchen ventilation:
https://ibb.co/4YVtZtk

Firmware 2.4.6
config:
{
"c": { "l": 13 },
"a": [{
"t": 1,
"0": { "r": [{ "g": 12, "v": 0 }] },
"1": { "r": [{ "g": 12, "v": 1 }] },
"b": [{ "g": 0 }],
"s": 0
}, {
"t": 24,
"g": 14,
"n": 4,
"j": 0.5,
"y1": [{
"v": 80,
"r": 1,
"0": { "r": [{ "g": 12, "v": 1 }] }
}, {
"v": 79,
"0": { "r": [{ "g": 12, "v": 0 }] }
}]
}]
}

Bathroom ventilation (setup like 3 weeks ago)
https://ibb.co/C1wbChb

Firmware 2.5.7
config:
{
"c": { "l": 13 },
"a": [{
"t": 1,
"0": { "r": [{ "g": 12, "v": 0 }] },
"1": { "r": [{ "g": 12, "v": 1 }] },
"b": [{ "g": 0 }],
"s": 0
}, {
"t": 24,
"g": 14,
"n": 4,
"j": 0.5
}]
}

£rd one I really needed it so I ended opening it and flashing it. it was under 1.x something version. It is now working fine on the new network, proves the network is ok.
Config:
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":22,"g":14,"n":3},{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]}]}

How can I change the wifi of the 2 others without getting into my ceiling with the laptop please?

Thanks

RavenSystem, 2020-11-09:
As I said, emergency setup mode works fine with all versions.

closed 2020-11-09 by RavenSystem

coyotanark, 2020-11-09:
As I already mentioned many times, nope it does not. And don’t advise to erase the device as you did on the other issue to pretend there is no problem