RavenSystem/esp-homekit-devices #1179: Eachen ST-DC2-TY ¡New Product!

STijeras91, 2020-10-18:
I found it in Amazon under the newsgoal brand, the truth is that prices are not bad and is a prime prosucto. Which I find interesting because you have it in 24h at home.

After I can flash this device after a few hours... Since it could not be done in the usual way on other devices with GPIO0. I have then managed to put it in Homekit with HAA. Now I have the worst and it is to identify the GPIO for the JSON configuration. I managed to find this on the Tasmota website. I need some help, the idea is that it works as follows. Thank you so much!

1 Lock Mechanism, 1 inching relay (0.6 sec both)

{"c":{"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12,"v":1}]},"1":{"r":[{"g":12,"v":0}]},"t":4,"i":0.6,"s":1,"b":[{"g":0,"t":1}]},{"0":{"r":[{"g":5,"v":0}]},"1":{"r":[{"g":5,"v":1}]},"i":0.6,"s":0,"b":[{"g":0,"t":2}]}]}

This is the board

[image]

[image]

tokamac, 2020-10-19:
This seems to be quite similar to a Sonoff SV, but with two relays instead of one. It accepts 5DC from USB or 7V-32V range either DC or AC as input power, can pass through 110-240 VAC from the mains, has inching, and there is a version adding RF in addition to Wi-Fi. I’ve found this diagram. Hope we can use it soon reliably with HAA.

Actually this a range of products: [DC-1](https://ewelink.eachen.cc/product/eachen-st-dc1/) (one relay); [DC-2](https://ewelink.eachen.cc/product/eachen-st-dc2/) (two relays); [DC-4](https://ewelink.eachen.cc/product/eachen-st-dc4/) (four relays)

[image]

STijeras91, 2020-10-19:
The only thing I'm going crazy setting up the JSON since I don't know the GPIO outputs. Let's see if anyone can help me with this.

tokamac, 2020-10-19:
According to [Tasmota template](https://templates.blakadder.com/eachen_ST-DC2.html), relays are apparently operated through GPIOs 12, 5, 3, 15. I don't know why they mention four relays instead of two for ST-DC-2. Maybe they meant ST-DC-4? Whatever, at first just try GPIO 12 (relay 1) and GPIO 5 (relay 2).

Also worth noting on the [Tasmota GitHub source page](https://github.com/blakadder/templates/blob/master/_templates/eachen_ST-DC2) for this device: 
> To flash the board leads will need to be soldered to 3.3v, RX, TX, Ground, GPIO 4, & GPIO 14.
Only leads Ground, GPIO 4, & GPIO 14 are needed to operate.

and
> [for a garage door] hook the NO side of the relays to the momentary button on the garage door. Use a reed switch wired between GPIO 4 and ground to signal power 3, and GPIO 14 and ground to signal power 4.

STijeras91, 2020-10-19:
It may be perhaps due to the two outputs that each NO, COM, NC relay has. The only thing I need is how to do the JSON for two switches with 1sec of Inching Time. But I can't find a way to make it work. And I already have it flashed and put in HAA and Homekit. Thanks!

tokamac, 2020-11-07:
The Tasmota page about the ST-DC1 (1 relay) is correct as this model has 1 button (+ 1 "mode" button), 1 switch (to connect a physical switch) and a single relay.

But the Tasmota page about the ST-DC2 (two relays) is mistaken, it is really about the ST-DC4 (four relays) which has 4 buttons (+ 1 "mode" button), 4 switches and 4 relays. And they are also mistaken saying one needs to connect GPIOs 4 and 14 in order to flash the board. Only Vcc/GND/Tx/Rx are required, as well as pressing Button #1 for a few secs while plugging the FTDI board, as usual (maybe these comments were from a former version of the board).

tokamac, 2020-11-07:
I bought Eachen ST-DC4-TY, the one with 4 black (not blue) relay modules, 2020 board written "tuya_2.2" on PCB. I managed to flash it with HAA 3.5.18 but like you I cannot find a JSON code that works.

The pairing button (GPIO 0) is the button 1 on the board.

I tested all possible GPIOs Tasmota says being available (except #13 which is for the LED of course): GPIOS 0, 4, 5, 9, 12, 14, 15, set as 7 independent switches. Although the JSON seems valid and is registered when applied, absolutely NO ONE works. Each switch in HomeKit only briefly triggers the blue LED on the board and no relay is triggered.

My code (doesn't work):
`{"c":{"l":13,"b":[{"g":0,"t":5}],"n":"test-focus"},"a":[{"0":{"r":[{"g":0,"v":0}]},"1":{"r":[{"g":0,"v":1}]},"t":1,"b":[],"s":0},{"0":{"r":[{"g":4,"v":0}]},"1":{"r":[{"g":4,"v":1}]},"t":1,"b":[],"s":0},{"0":{"r":[{"g":5,"v":0}]},"1":{"r":[{"g":5,"v":1}]},"t":1,"b":[],"s":0},{"0":{"r":[{"g":9,"v":1}]},"1":{"r":[{"g":9,"v":1}]},"t":1,"b":[],"s":0},{"0":{"r":[{"g":12,"v":0}]},"1":{"r":[{"g":12,"v":1}]},"t":1,"b":[],"s":0},{"0":{"r":[{"g":14,"v":0}]},"1":{"r":[{"g":14,"v":1}]},"t":1,"b":[],"s":0},{"0":{"r":[{"g":15,"v":0}]},"1":{"r":[{"g":15,"v":1}]},"t":1,"b":[],"s":0}]}`

After several attempts to switch the relays on and off with HomeKit, the switches in Home.app become flagged with a "?" before being available again later.

The other weird thing to note is that the physical buttons 1, 2, 3, 4 on the board continue to activate the corresponding relays, despite being not declared in the JSON. And the 5th "mode" button keeps cycling the various pre-programmed modes (normal, inching, interlock) despite the memory being erased by esptool. I suspect the board to have another eeprom interfering with HAA and preventing it to properly activate the relays (?).

STijeras91, 2020-11-09:
I would like to know if you can really work a code in HAA for these devices. Since I have been searching like crazy everywhere and in the end I have had to give it up. Thanks

StofLE, 2020-11-25:
Did you try Tasmota? Maybe easier to find correct gpios.

closed 2020-12-25 by stale[bot]