RavenSystem/esp-homekit-devices #1186: shelly garage door stops working since 3.5.10

lickonline, 2020-10-23:
my shelly garage door stops working since (about) release 3.5.10. 
using the json "Garage Door Example" in the wiki, the gpio 4 is used to open/close garage door it activates the output, with an reset time of 0,5s. (relay on for 0,5s, then off)
since 3.5.10 initial the relay goes on, but not off. in further use it goes TWICE on/off (relay on for 0,5s, then off .. and again relay on for 0,5s, then off) , than means garage door opens, and get immediately stop command. (door moves only for 2 inch)

any ideas?  (am I the only one with this issue?)
________________________________________________

PS: when I flash older e.g. 3.5.3, it updates automatically to newest 3.5.12, even thoug "automatic ota update" is not activated).
also when I attempt to downgrade to to an older release, via local server, shelly does not connect anymore to local wifi 

is there any other way, to get an older release working?

macjl, 2020-10-23:
I think you are facing this issue : 
https://github.com/RavenSystem/esp-homekit-devices/issues/1173

i3laze, 2020-10-25:
Indeed garage behavior has changed recently. My two doors, Sonoff SV and Basic sometime now become unresponsive to original remote, as if HAA leaves the garage button circuit closed (don’t release the inch) for some reason. Will investigate.

lickonline, 2020-10-25:
thank you macjl! 
issue is similar, but found no solutions for me, till now

(by the way: I know that I am able to process shelly/flashing/homekit integration, because yesterday I flashed release 3.5.13 with
a json for a "Shelly 1 switch as electric door mechanism trigger" ..... and it works perfect.)

in the case of shelly for garage door with ravensystem ... now I gave up (to many hurdles all over) (a mess with open end, spend too many time with it)
I purchased "meross smart garage for homekit " and (with a modification of door opening time in meross app) it works perfect
(it works perfect now, no wasted time here)

RavenSystem, 2020-11-09:
Duplicated: https://github.com/RavenSystem/esp-homekit-devices/issues/1173

closed 2020-11-09 by RavenSystem