RavenSystem/esp-homekit-devices #1188: How to keep older release, avoid OTA automatic OTA update? / How to downgrade OTA release ?

lickonline, 2020-10-23:
when I flash older release e.g. 3.5.3, it updates automatically to newest 3.5.12, even thoug I do not make checkbox auto OTA update
in which way is it possible to keep an olderrelase?

also tried to downgrade o an older release, filled the form for an local server for custom updates, but after that shelly will not connect anymore to local wifi. (if I leave the form empty, then shelly does connect to local wifi)
any idea how to downgrade otherwise?

komarofleo, 2020-10-23:
You can flash via this command (mac) on terminal
esptool.py -p /dev/cu.usbserial-1A110 erase_flash

then 
esptool.py -p /dev/cu.usbserial-1A110 --baud 115200 write_flash -fs 1MB -fm dout -ff 40m 0x0 /Users/yourcomp/355/rboot.bin 0x1000 /Users/yourcomp/355/blank_config.bin 0x2000 /Users/yourcomp/355/haamain.bin

in 355 folder you need place this files from release you want:
blank_config.bin
fullhaaboot.bin
haa_downloader.sh
haaboot.bin
haaboot.bin.sec
haamain.bin
haamain.bin.sec
haaversion
otamain.bin
otamain.bin.sec
otaversion
rboot.bin

lickonline, 2020-10-25:
thank you komarofleo! 
I tried this way, and initial it works, but in the last step there is no homekit device found

(by the way: I know that I am able to process shelly/flashing/homekit integration, because yesterday I flashed release 3.5.13 with
a json for a "Shelly 1 switch as electric door mechanism trigger" ..... and it works perfect.)

in the case of shelly for garage door with ravensystem ... now I gave up (to many hurdels all over) (a mess with open end, spend too many time with it)
I purchased "meross smart garage for homekit " and (with a modification of door opening time in meross app) it works perfect
(it works perfect now, no wasted time here)

closed 2020-11-24 by stale[bot]