RavenSystem/esp-homekit-devices #1193: change in garage door JSON in last version

pilot1981, 2020-10-26:
Hi,
I see you change something related to initial state of relay in last firmware...

I already use this JSON for my garage door:

{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":40,"d":18,"f3":[{"g":3,"t":0}],"f4":[{"g":3}],"0":{"r":[{"g":12,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":12,"i":1.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},"3":{"a":2},"b":[{"g":0}]},{"1":{"s":[{"a":1}]}},{"i":0.5}]}

How I can change this JSON to adapt it for vastest firmware versions?

RavenSystem, 2020-10-26:
{"c":{"l":13,"b":[{"g":0,"t":5}],"0":{"r":[{"g":12}]}},"a":[{"t":40,"d":18,"f3":[{"g":3,"t":0}],"f4":[{"g":3}],"0":{"r":[{"g":12,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":12,"i":1.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},"3":{"a":2},"b":[{"g":0}]},{"1":{"s":[{"a":1}]}},{"i":0.5}]}

pilot1981, 2020-10-26:
thanks a lot, I go to try it today!

I can try this JSON with older firmware (3.5.2)?

...or before I have to update to 3.5.13 and then change JSON?

I ask this only to understand if I have and alternative solution in case new JSON don't work...

morphix9, 2020-10-27:
@RavenSystem can you help me with this one? (I'm afraid to update) 

{"c":{"l":13,"o":3,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":4,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":4,"i":1.5},{"g":4,"v":1,"i":0.5},{"g":4,"v":1,"i":2}]},"3":{"a":2},"t":40,"d":18,"f3":[{"g":3,"t":0}],"f4":[{"g":3,"t":1}],"s":0,"b":[]},{"1":{"s":[{"a":1}]},"i":0.5,"b":[],"s":0}]}

RavenSystem, 2020-10-27:
@morphix9 
`{"c":{"l":13,"o":3,"b":[{"g":0,"t":5}],"0":{"r":[{"g":4}]}},"a":[{"0":{"r":[{"g":4,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":4,"i":1.5},{"g":4,"v":1,"i":0.5},{"g":4,"v":1,"i":2}]},"3":{"a":2},"t":40,"d":18,"f3":[{"g":3,"t":0}],"f4":[{"g":3}]},{"1":{"s":[{"a":1}]},"i":0.5}]}`

RavenSystem, 2020-10-27:
@pilot1981 You can use the new JSON with old firmware.

morphix9, 2020-10-27:
So the change is that "0":{"r":[{"g":4}]}} has to be added to the JSON? @RavenSystem

RavenSystem, 2020-10-27:
> So the change is that "0":{"r":[{"g":4}]}} has to be added to the JSON? @RavenSystem

Yes, to set initial relay state to LOW.

pilot1981, 2020-10-28:
JSON works fine. thanks a lot @RavenSystem.

I also go to update tested JSON issue!

closed 2020-10-28 by pilot1981