RavenSystem/esp-homekit-devices #1194: Sonoff Mini Dry Contact as a Garage Door?

mestregroda, 2020-10-26:
Hello,
I've being using Sonoff Mini (Tasmota) as a switch in my garage door, for a while. I modified the sonoff into a dry contact relay.
It is working flawslessly, both my remote control and when using homekit (Raspberry/Homebridge) works as it should. I am able to use Siri through iOS shortcuts (Open the Gate / Close the Gate). S1 and S2 are not used.
But I am wondering If I can go some steps further. For example, I am wondering if upgrade from tasmota to fullhaaboot.ini, I can transform sonoff in a native homekit device and then, maybe, I can not only control the gate as before, but also as a "garage door" instead a "switch", and finally I could control the gate using Carplay also.
What do you think? Is it feasible?

mestregroda, 2020-10-27:
So... I figured it out how to solve several issues. Now I am able to use sonoff mini as dry contact and as a "garage door" instead a "switch". I can ask Siri, to open/close the gate. Apple Carplay shows Gate button and works as supposed. Home app works as supposed. I can even press button on sonoff for open/close gate. Everything works just fine, except for one thing... my remote control works only before any homekit action. And I really need the remote control working together with homekit.
I dont know what else I can do, I am not a tech person and my skills are very limited in this particular area. Below are the sonoff picture with dry contact, gate board picture (button/ground) and my Jsonf settings:
[image]
[image]
Json Configurator:
{
  "c": {
    "l": 13,
    "b": [
      {
        "g": 0,
        "t": 5
      }
    ]
  },
  "a": [
    {
      "0": {
        "r": [
          {
            "g": 12,
            "v": 0,
            "i": 0.5
          }
        ]
      },
      "1": {
        "r": [
          {
            "g": 12,
            "v": 0,
            "i": 0.5
          }
        ]
      },
      "t": 40,
      "s": 5,
      "d": 14,
      "b": [
        {
          "g": 0,
          "t": 5
        }
      ],
      "f0": [],
      "f1": []
    }
  ]
}

And finally, the remote control....lol
[image]
Any help will be very very apprecciated.

mestregroda, 2020-10-28:
@RavenSystem Your thoughts are always welcome.

RavenSystem, 2020-10-28:
I understand that your issue is when you use your remote control, garage door dones't change its status in Homekit, no?

mestregroda, 2020-10-28:
> I understand that your issue is when you use your remote control, garage door doesn't change its status in Homekit, no?
Let's say gate Homekit icon is greyed with status "Gate Closed", then I can press remote control button, and the gate will open as supposed.
But after that, remote control button wont work anymore. Gate wont move 
I can use siri, or carplay button, or home app, to close the gate. 
The only issue that remains is how to make remote control work together with homekit, Even if I was able to open the gate with remote control and garage door doesn't change its status in homekit I would be very satisfied. I dont know if I was clear in my explannation. My english is trash.

mestregroda, 2020-10-28:
The following Json works with remote control:
{
  "c": {
    "l": 13,
    "b": [
      {
        "g": 0,
        "t": 5
      }
    ]
  },
  "a": [
    {
      "0": {
        "r": [
          {
            "g": 12,
            "v": 1,
            "i": 0.5
          }
        ]
      },
      "1": {
        "r": [
          {
            "g": 12,
            "v": 1,
            "i": 0.5
          }
        ]
      },
      "2": {
        "r": [
          {
            "g": 12,
            "v": 1,
            "i": 0.5
          }
        ]
      },
      "3": {
        "r": [
          {
            "g": 12,
            "v": 1,
            "i": 0.5
          }
        ]
      },
      "4": {
        "r": []
      },
      "t": 40,
      "s": 5,
      "d": 14,
      "b": [
        {
          "g": 0,
          "t": 0
        }
      ],
      "f0": [],
      "f1": []
    }
  ]
}

But when I use remote control, status door opened/door close doesn't change. Should I install a magnetic senson on the gate/door?

KS-Epoch, 2020-10-28:
Are you sure that tasmota was working as on/off switch? Maybe it had unlatched mode?

I have same issue with my gates automation. My gates have normal closed input: OPEN. When my relay opens it, no interaction possible with remote.

mestregroda, 2020-10-28:
I have a modified Sonoff Mini (dry contact). First: dry contact, second: tasmota firmware, third: fullhaaboot.bin
It is working as garage door, but as a switch it works too. The Json code for switch is:
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1,"i":0.5}]},"b":[{"g":0}],"s":0,"i":0.5}]}

This Json works with my remote control too.

closed 2020-11-28 by stale[bot]

sfaxman, 2021-01-11:
@mestregroda : Would you please explain how did you modified the sonoff mini into a dry contact relay?
Thank you

regissg, 2021-03-04:
@mestregroda : I am also interested to know how you manage to modify the sonoff mini into a dry contact relay?

guilauriano, 2021-08-22:
Se você usar um contador por atalho do iOS, ao executar o atalho, ele verifica se portão true ou false e inicia um contador.

fiz isso para usar o back tap para controlar o sonoff, visto que o ewelink não fornece a opção de apenas reverter

outra opção seria captar a notificação do ewelink atraves de ifttt, enviar email ou algo assim, e por atalhos, somar no contador para identificar se está aberto ou fechado.

qosmio, 2022-01-06:
> So... I figured it out how to solve several issues. Now I am able to use sonoff mini as dry contact and as a "garage door" instead a "switch". I can ask Siri, to open/close the gate. Apple Carplay shows Gate button and works as supposed. Home app works as supposed.

These type of replies are so frustrating. 

"I am looking for help on XYZ... never mind, I figured it out."

Why even reply? Two people have asked you to please explain what it was you did to convert it to a dry contact switch. You haven't replied with any updates so I assume you also figured the "programming" question you had.

guilauriano, 2022-01-06:
Ola, vou te ajudar fique tranquilo

Enviado do iPhone

sudosoul, 2022-03-27:
@mestregroda interested in knowing how you converted it to a dry contact relay

cristidobrescu, 2022-04-20:
This is also applied on Sonoff Mini R2, just follow this
https://youtu.be/vsVtF_6RrTM

mmaday, 2023-05-31:
> @qosmio @sudosoul This is also applied on Sonoff Mini R2, just follow this https://youtu.be/vsVtF_6RrTM

I can confirm that the procedure @cristidobrescu linked works on a newer ZBMINI (v2.0 printed on the circuit board).  I also found the video https://youtu.be/JbxqXB5qfjQ helpful.