RavenSystem/esp-homekit-devices #1196: Help with HAA json and shelly 1

Alex-gpio, 2020-10-29:
Please I need help

I've this json string

lock mechanism with external push button (default closed)

{"a":[{"t":4,"i":3,"1":{"r":[{"g":4}]},"0":{"r":[{"g":4,"v":1,"i":0.5}]},"b":[{"g":5,"p":0,"i":1,"t":0}]}]}

but i'need that the lock mechanism works to inverse mode (default open)
I need that the mechanism must be in opened state by default  (not closed as the code I gave)
for the rest of code as inching time 3 sec is correct.
moreover I need that when it has a power loss it restart unlocked or to previous state

Thanx a lot to everybody

gerarddvb, 2020-10-30:
{"a":[{"0":{"r":[{"g":4,"v":1,"i":0.5}]},"1":{"r":[{"g":4}]},"t":4,"i":3,"b":[{"g":5,"p":0,"i":1,"t":0}],"s":0}]}
This should make it when power is lost it boots to unlocked.

{"a":[{"0":{"r":[{"g":4,"v":1,"i":0.5}]},"1":{"r":[{"g":4}]},"t":4,"i":3,"b":[{"g":5,"p":0,"i":1,"t":0}],"s":5}]}
This should make it when power is lost it boots to previous state.

Hope this helps. You can paste your json lint on the bottom of this site and start changing;
https://glumb.github.io/haa-configurator/

Alex-gpio, 2020-10-31:
Thanx Gerarddvd!!

I tried your json code, unfortunately it doesn’t work on shelly v1.
The device boots correctly and it connects to my wifi, but homekit doesn’t find it, so is not possible to add the device in the apple home app.

I’m using the latest HAA firmware.

gerarddvb, 2020-10-31:
Did you reset the Homekit ID in setup?

Alex-gpio, 2020-10-31:
Of course!

gerarddvb, 2020-10-31:
I have this problem too. Sometimes homekit doesnt want to reconise it. Restarting router fixes it for me

Alex-gpio, 2020-10-31:
Good idea I’ll try this solution

Alex-gpio, 2020-10-31:
Nothing, it doesn’t work

Alex-gpio, 2020-11-01:
It works!
I don’t know why. But leaving the device connected after 3 hours it become discoverable by homekit.

Bu now I need help.

I need that the lock would be in opened state by default and non closed.

This is the json I used, how can I set it for shiwing the lock opened?

{"a":[{"0":{"r":[{"g":4,"v":1,"i":0.5}]},"1":{"r":[{"g":4}]},"t":4,"i":3,"b":[{"g":5,"p":0,"i":1,"t":0}],"s":5}]}

closed 2020-12-01 by stale[bot]