RavenSystem/esp-homekit-devices #1198: Update

TheReiner, 2020-11-02:
Hello, I have about thirty devices and it takes a long time to update ... Take them one by one, switch off and on 8 times, find your IP, start in configuration mode, start the update, and check that everything is OK.
Could there be a faster way? A batch update?
Or quite simply a Web server on permanently, to only make the updates.

Thanks.

peros550, 2020-11-02:
You can modify your json to show one additional switch for every device that you have. That (virtual) switch could be configured to either trigger the setup mode or even better for your case, to trigger the update process. 

Since you have 30 devices, if you follow that route you would end up with additional 30 switches in your home app. The workaround idea to minimize the hassle is to add them all together to a new room (i.e. the update room) and then probably setup a scene that will activate all these switches at once.

TheReiner, 2020-11-02:
Thanks Peros550... 
It's a good idea, but not sure it's a good one with so many devices. 30 switches and 30 virtual switches, that's 60 switches, not counting the rest of my house!
it's not possible to have some things like that ... http: //x.x.x.x/Update

pilot1981, 2020-11-02:
I have about 40 devices, I followed suggestion of @peros550 and grouped all dummy switches in only one dummy switch to switch on only when I would update all my devices at same time with on one touch...amazing solution!

macjl, 2020-11-02:
I'm using same kind of solution on ~20 devices.
- Added 1 dummy switch that activate the Setup Mode on each devices
- Enable the 'Auto OTA Update'
- Group all this dully switchs in one switch in Home App

So, when I want to update all, I just have to switch on the dummy switch, and 90 seconds later, the updates begins.

TheReiner, 2020-11-02:
Thanks,
Ok i will do that... But 
Where to create a dummy or Virtual Switch ? i don't find in the wiki...

"s": [{ "a": 1}] Setup Mode
"s": [{ "a": 2}] Update.

This is my configuration for a Sonoff Basic

{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0},{"g":4},{"g":4,"t":0}]}]}

peros550, 2020-11-02:
It would be a nice idea if HAA could expose some sort of API, so that other programs could check, control and possibly give a command to update the devices without adding virtual switches.

Hanimaniac, 2020-11-02:
Hey Friends, 
when you have 30 or more devices in your wifi, which wifi setup do you use? I have a Fritzbox and a Netgear Orbi System, but I have the Feeling, that my Devices (around 25) are to much and I often get a no response from time to time by different devices.
Greets Hanimaniac

macjl, 2020-11-02:
@TheReiner 
For example, with a setup dummy switch
`{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0},{"g":4},{"g":4,"t":0}]},{"t":1,"1":{"s":[{"a":1}]}}]}`

@Hanimaniac 
I had same kind of problem with many devices on my network. I've created a separated Wifi network dedicated to devices which doesn't need high bandwith (Homekit, printers, etc...), and an other one for user's devices (iPhone, iPad, computers, etc...). The two networks are managed by different access points.
Till this separation, no more problems for me.

rjblake, 2020-11-03:
@macjl - or if you have a managed network you could add these into their own VLAN and use separate SSID (if your access point/router supports this)

macjl, 2020-11-03:
@rjblake 
Even if you use separate SSID, you still have same frequencies by using just one AP.
With different AP, you can have multiples frequencies bands. The performances are much better with this approch in my home.

rjblake, 2020-11-04:
@macjl - indeed, as you state. If you life in a densely populated area (such as I do), the 2.4GHz channels can be heavily populated in which case you may struggle to find non-overlapping frequencies (if at all); which would lead to other issues. @Hanimaniac has the Netgear Orbi, which should be able to support up to 250 clients without issue (especially since the IoT devices are low traffic and traffic likely in the order of kb's under normal operation); so I wouldn't expect 30 or so devices to be an issue; but cannot speak to the performance of the Orbi mesh performance

TheReiner, 2020-11-17:
@macjl Thanks It works perfectly, and you were all right, with a room where we put all the devices in it :) Very fast to update everyone ^^

closed 2020-12-17 by stale[bot]