RavenSystem/esp-homekit-devices #1199: Magic Home RGBW initial state

DanielMosco, 2020-11-03:
Hello all. Need help. I use the HAA flashed Magic Home controller for the last half a year and i’m happy except that every time i have a power outage, the device stays off after it reboots. I used JSON:
{"c":{"q":850,"b":[{"g":0,"t":5}]},"a":[{"t":30,"r":15,"g":13,"v":12,"w":14,"b":[{"g":0}]}]} 
I cannot figure out how to add “s” to this thing. I need to add S=5. 
Please help to edit the JSON if it possible. 
And, how to reset this device without reflashing? 
Thanks in advance, Daniel

mafyata, 2020-11-03:
{"c":{"q":850,"b":[{"g":0,"t":5}]},"a":[{"t":30,"s":5,"r":15,"g":13,"v":12,"w":14,"b":[{"g":0}]}]}

DanielMosco, 2020-11-03:
Thanks. So simple 🤦🏻‍♂️ 
Do know how to reset the device so i can replace the json without flashing it again?

mafyata, 2020-11-03:
Enter the device in setup mode, search for it’s wifi and connect to it.

DanielMosco, 2020-11-03:
I possibly did not explain myself well. 
How do i enter setup mode on Magic Home controller. There are no physical button to do that. 
Thanks for your help

wsc96, 2020-11-13:
From the Wiki:

Emergency Setup Mode: If a device has its power cut or freezes within 3 seconds of boot, next time it boots it will go directly to setup mode.

Hoefnix, 2020-11-29:
> From the Wiki:
> 
> Emergency Setup Mode: If a device has its power cut or freezes within 3 seconds of boot, next time it boots it will go directly to setup mode.

...or switch it on / off eight times in the homekit app

closed 2020-11-30 by DanielMosco