RavenSystem/esp-homekit-devices #1200: MagicHome RGB Controller Issue

epicsepi, 2020-11-03:
I have 3 same MagicHome RGB controller, and 3-4 different LED strip. I tried some json, but doesnt working all color in one time. 
[image]
[image]
[image]

Red doesnt lightning, why?

wayn3, 2020-11-03:
Guessing the config for the red led is setup with the wrong gpio?

epicsepi, 2020-11-03:
Its possible. But, i dont know what I have to do. Where Can I change the gpio?

wayn3, 2020-11-04:
In your json config, post it here and I'll show you

wayn3, 2020-11-04:
Taking a guess, this is what I would expect (including a virtual switch to enter setup mode)
```
{
	"c":{
		"n":"MagicHomeRGB"
	},
	"a":[{
		"t":30,
		"r":5,
		"g":12,
		"v":13
	},{
		"1":{"s":[{"a":1}]},
		"i":0.5
	}]
}

epicsepi, 2020-11-04:
{"c":{"q":850,"b":[{"g":0,"t":5}]},"a":[{"t":30,"r":15,"g":13,"v":12,"b":[{"g":0}]}]}

I saw this in device database. Thank you very much!!!

epicsepi, 2020-11-04:
Ok, its working. Red is red, blue is blue, green is green, but white (middle of the color scale) is not close to white. Its light purple. What I have to change in the json, to get true white? Thanks!!!! 

this is the json, what you told: {"c":{"n":"MagicHomeRGB"},"a":[{"t":30,"r":5,"g":12,"v":13},{"1":{"s":[{"a":1}]},"i":0.5}]}

wayn3, 2020-11-04:
RGB strips never really produce a great white light, often ends up blue~ish, you could try setting factors but I haven’t use this functionality

https://github.com/RavenSystem/esp-homekit-devices/wiki/Lightbulb#RGBW-Factor

epicsepi, 2020-11-04:
Okey, no problem!! Thank you all!! You saved my life!!🙌🙌😋

closed 2020-11-09 by RavenSystem