RavenSystem/esp-homekit-devices #1201: Bug: GPIO input detection stop working after HAA 3.5.7 to 3.5.8 update.

vsevolod-volkov, 2020-11-05:
**Devices**: more than ten Shelly 1s affected.
**Symptoms**: after applying HAA 3.5.8 update device stops detecting GPIO 5 input at all neither for action/state change nor for setup mode detection. Some device worked fine just after update but definitely stop working after execution several HomeKit-initiated actions.
**Affected HAA versions**: 3.5.8, 3.5.9, 3.5.16 and 3.5.17 (3.5.10-3.5.15 were not tested).
**Work-around**: downgrade to HAA 3.5.7.

**JSON configuration**: There was two configuration kinds on devices (both kinds were affected):
1) switch:
{
  "c": {
    "n": "Light",
    "o": 3,
    "b": [{ "g": 0, "t": 5 }]
  },
    "a": [
        {
            "0": {
                "r": [
                    {
                        "g": 4
                    }
                ]
            },
            "1": {
                "r": [
                    {
                        "g": 4,
                        "v": 1
                    }
                ]
            },
            "f0": [
                {
                    "g": 5,
                    "t": 0
                }
            ],
            "f1": [
                {
                    "g": 5,
                    "p": 0
                }
            ]
        }
    ]
}

2) switch with auto-off timer:
{
  "c": {
    "n": "Light with auto-off",
    "o": 3,
    "b": [{ "g": 0, "t": 5 }]
  },
    "a": [
        {
            "0": {
                "r": [
                    {
                        "g": 4
                    }
                ]
            },
            "1": {
                "r": [
                    {
                        "g": 4,
                        "v": 1
                    }
                ]
            }
        },
        {
            "d": 1800,
            "0": {
                "m": [
                    {
                        "g": 1,
                        "v": 0
                    }
                ]
            },
            "1": {
                "m": [
                    {
                        "g": 1,
                        "v": 1
                    }
                ]
            },
            "f0": [
                {
                    "g": 5,
                    "t": 0
                }
            ],
            "f1": [
                {
                    "g": 5,
                    "p": 0
                }
            ]
        }
    ]
}

vsidlu, 2020-11-06:
Just to confirm that I also have this problem. Downgrade to 3.5.7 fix problem. I have sonoff basic with button on gpio14. Logs do not show problem. After some time gpio14 button stop to working.

RavenSystem, 2020-11-06:
Please, try v3.5.18 and report.

vsevolod-volkov, 2020-11-06:
Looks like it is fixed. Thank a lot, Jose!

closed 2020-11-06 by vsevolod-volkov