RavenSystem/esp-homekit-devices #1202: remote sensor

pixidixi88, 2020-11-05:
Hello,
I have a problem with the temperature sensors, after some time (few hours) they are not providing temperature and I see 0.0 temperature in home app. I need to unplug and plug again to work it again.
I have checked different Sonoff devices (basic and TH10 TH16) and different sensors (DHT22 and DS18B20) and it is always the same. I need to say that they have quite long cables (5 - 10m) I suppose it is a problem because if I connect sensor directly to Sonoff it works well without such issues. 
Is it possible to have remote sensor for Sonoff? or is there any chance to use sensor connected in one device as a temperature input for another (thermostat)?
Electric valves are located in one room in the central box with sonoff's but sensors are in the different rooms, so I cannot shorter the cables.
Has anyone experienced similar problems?

vsidlu, 2020-11-05:
hi,
I use esp8266 with dht22 as thermostat and sonoff 4ch as switch for heating floor heads. Then I use thermostat to send https://github.com/RavenSystem/esp-homekit-devices/wiki/Accessory-Configuration#Send-HTTP-Request-Actions
to on or off 4ch.

pixidixi88, 2020-11-05:
Thank you! could you share json for both devices? I am not IT specialist. everything I am learning by myself from scratch.

vsidlu, 2020-11-05:
the same story! I am not IT spec at all.

for 4ch - 
{"c":{"l":13,"u":1,"n":"boiler_1","b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":15}]},"1":{"r":[{"g":15,"v":1}]},"b":[{"g":14}]},{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]},{"0":{"r":[{"g":5}]},"1":{"r":[{"g":5,"v":1}]},"b":[{"g":9}]},{"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":10}]}]}

for esp8266 with dht22
{"c":{"u":1,"l":2,"n":"t_guestroom","i":1,"b":[{"g":0}]},"a":[{"t":25,"s":5,"g":14,"0":{"h":[{"h":"10.0.1.8","p":5556,"m":1,"u":"characteristics",
"c":"{\"characteristics\":[{\"aid\":2,\"iid\":9,\"value\":false}]}"}]},"1":{"h":[{"h":"10.0.1.8","p":5556,"m":1,"u":"characteristics",
"c":"{\"characteristics\":[{\"aid\":2,\"iid\":9,\"value\":false}]}"}]},"3":{"h":[{"h":"10.0.1.8","p":5556,"m":1,"u":"characteristics",
"c":"{\"characteristics\":[{\"aid\":2,\"iid\":9,\"value\":true}]}"}]},"5":{"a":0}},{"i":0.5,"1":{"s":[{"a":1}]}}]}

you will need to fix 4ch ip in dhcp - 10.0.1.8 is 4ch ip.

closed 2020-12-05 by stale[bot]