RavenSystem/esp-homekit-devices #1203: motion sensors hang

ur5zeb, 2020-11-07:
Hi! After a while, motion sensors hang at me. The setting switch is working, the temperature sensor is working. After reboot everything works. Wemos D1 mini. pull-up 4.7k.
One sensor https://www.obsoleteradionics.com/srpg-2n-crow-combination-pir-and-glass-break-detector.html
The second RF433 sensor is a doorbell receiver. Connected via a NPN transistor switch
My Json
{
  "c": {"l": 2,"o": 3},
  
  "a": [{
        "t": 12,
        "i": 10,
        "f1": [{"g": 12,"p":0," i ": 1," t ": 0}]
    },{
        "t": 12,
        "i": 10,
        "f1": [{"g": 13,"p":0," i ": 1," t ": 0}]
    },{
        "1": {"s": [{"a": 1}]},"i": 0.5
    },{
        "t": 24,
        "g": 14,
        "n": 1,
        "j": 30
    }]
}

RavenSystem, 2020-11-07:
What HAA version are you using?

ur5zeb, 2020-11-08:
version 3.5.15

RavenSystem, 2020-11-08:
It is fixed in last version (v3.5.18).

closed 2020-11-08 by RavenSystem