RavenSystem/esp-homekit-devices #1204: Support Request: Sonoff ZBBridge

Be2daErtin, 2020-11-09:
Hi guys,

Not sure if this might be a good idea. But I think it will be great to support the ZBBridge from Sonoff?

Link:
https://www.itead.cc/sonoff-zbbridge.html

Some work already been done:
https://github.com/arendst/Tasmota/issues/8583

I think its supported by Tasmota:

https://tasmota.github.io/docs/Zigbee/

peros550, 2020-11-09:
It would be a great idea to support a ZigBee bridge. Imagine how many new devices could be added based on ZigBee protocol...

jakudo, 2020-11-25:
yes, would be great to be able to flash it with HAA somehow

Kristian8606, 2020-11-25:
Does anyone know what is the communication between zigbee radio module and esp

MattWestb, 2020-12-13:
The radion is integrated on the EFR32 chip but the protocol to the hest side is described and many more docks [UG100: EZSP Reference Guide](https://www.silabs.com/content/usergenerated/asi/cloud/attachments/siliconlabs/en/community/wireless/zigbee-and-thread/forum/jcr:content/content/primary/qna/debug_log_devicejo-lyxK/ug100-ezsp-reference-guide.pdf).

closed 2021-01-13 by stale[bot]