RavenSystem/esp-homekit-devices #1207: How to build a log of temperatures throughout the day (using Sonoff TH10)?

tlagarto, 2020-11-10:
Hi guys,

I'm not an IT person but managed to flash and pair around 30 sonoff devices to my Home app (no other hubs or servers besides an apple TV, a Mac, and an iPhone).
Some of those are temperature sensors and thermostats.
I would like to have the temperature readings of different rooms along the day so that I can understand how to optimize the  heating system usage.

How can I get temperature readings every 15min from each sensor and log that into a CSV file (or table or database)?
Do I need extra hardware to set this up?
How do I set it up?

(please remember I'm not a specialist and everything I did was based on tutorials and google searchers.. :P )

Thank you a lot in advance.

axxxkh, 2020-11-11:
You can start homebridge on your Mac. Then you’ll need to set up a trigger. For example plug-in for homebridge (homebridge automation switches) it will activate motion switch every X seconds (you can set it up in plugin config or eve app) and then set automation on this trigger to write data on your Mac over ssh. I don’t know how to do this with CSV. But I’m use influx database and Grafana to visualize data. All of this runs on RPi 2, but you can use your Mac (it should be active all the time )

peros550, 2020-11-11:
There is an online *free service by thingspeak.com. If your don't mind your data being send there you may use it. You 'll have to register for an account and then create a channel and field(s) that will hold your different type of measurements. For example,  one channel could be for a single house and multiple fields for your sensors.  Your HAA device should send measurements in the form of http requests. Therefore, every i.e. 15 minutes your HAA device should make a request like this: 

https://api.thingspeak.com/update?api_key=**TC2ZTRQNNHRRH35N**&field1=**30.5** 

where:

TC2ZTRQNNHRRH35N should be your write API, you will be given a personal API upon creating your channel
30.5 = is the value of your sensor. I randomly choose 30.5 

Thingspeak has a free ios app where you can visualize measurements. 

For what it costs, I found it pretty descent. And there are more possibilities (for complex analysis, visualizations) that are beyond my capabilities.   

*The thingspeak service is practically free if your needs are bellow 8.200 messages per day. If you plan to post every 15minutes (15x24=360)  you are well below that limit.

Hoefnix, 2020-11-29:
> There is an online *free service by thingspeak.com. If your don't mind your data being send there you may use it. You 'll have to register for an account and then create a channel and field(s) that [...]

At the moment i'm reading humidity values through the rest api with a python script and write the measurements to thingspeak (and use them for switching central ventilation) I never got it working they way you suggest. my current json is {"c":{"n":"dht22","u":1},"a":[{"t": 24, "g": 2, "n": 2}]}  you help me to incorporate the call to thingspeak every nth second/minute?

closed 2020-12-29 by stale[bot]