RavenSystem/esp-homekit-devices #1208: Factory reset, wifi change

arturlukosek, 2020-11-11:
I do have a sonoff basic with dry contact and door sensor to open and close the gate. Since I have changed the router cant put the sonoff back to the mode I can set the new wifi password. 
How to reset on the device to get to the wifi configuration settings?

rbrtsluk, 2020-11-11:
> I do have a sonoff basic with dry contact and door sensor to open and close the gate. Since I have changed the router cant put the sonoff back to the mode I can set the new wifi password.
> How to reset on the device to get to the wifi configuration settings?

Hey mate can I do this with a basic r3?
Do you have a tutorial you went by or you just know what your doing?

xrust83, 2020-11-29:
8 times quickly turn on turn off the relay, then go to the IP in the browser

closed 2020-12-30 by stale[bot]