RavenSystem/esp-homekit-devices #1214: Howto control another HomeKit devices using HAA

pilot1981, 2020-11-15:
Hi,
My next project is to use traditional wall switch with Shelly 1 or 2.5 to control a load with smart led bulb coloured (ex. Philips hue, eve, ...).

I would have a Shelly switch attached to wall switch that work in this way:
- if someone switch on wall button attached to Shelly with HAA, HAA send command to smart led bulb to switch on;
- if someone switch off wall  button attached to Shelly with HAA, HAA send command to smart led bulb to switch off;

In this way I can use traditional switch and use also smart led bulb with different colour and adaptive feature

Hanimaniac, 2020-11-16:
Hey Pilot,
if I understand Your Problem, you kann build an automation in Homekit e.g. when the HAA Shelly is turned ON, then turn on the LED Bulb (which must be also in Homekit). 
With that you can switch a light with yor wall switch without a physically connection.

pilot1981, 2020-11-16:
Yea, I can use HomeKit automation, but I prefer more speed...so I thinked a specific JSON in HAA Shelly Could be better...

pilot1981, 2020-11-24:
@RavenSystem no any idea? I remember already see something useful in the past developing, but I'm not able to find it..or if it's not possibile now, I ask if it can be possibile with future releases...

Hoefnix, 2020-11-24:
I think you mean this https://github.com/RavenSystem/esp-homekit-devices/wiki/Accessory-Configuration#actions 
If your bulb can be managed through a http call then specifically this sections might be of interest: https://github.com/RavenSystem/esp-homekit-devices/wiki/Accessory-Configuration#send-http-request-actions

If the bulb is flashed with HAA you can enable the homekit REST API https://github.com/RavenSystem/esp-homekit-devices/wiki/General-Configuration#enable-homekit-rest-api

pilot1981, 2020-11-24:
great answer @Hoefnix! thanks very much for help!

You also know any commercial light bulb homekit compatible that support HTTP call? PHILIPS HUE?

For now I found this material about PHILIPS HUE:

https://community.zipato.com/topic/phillips-hue-basic-http-control

https://blog.kloud.com.au/2018/03/19/commanding-your-philips-hue-lights-with-powershell/

https://developers.meethue.com/develop/get-started-2/

...«It – could – work!»...

:-)

https://github.com/tigoe/hue-control

Hoefnix, 2020-11-24:
> great answer @Hoefnix! thanks very much for help!
>
> You also know any commercial light bulb homekit compatible that support HTTP call? PHILIPS HUE?
>
> For now I found this material about PHILIPS HUE: [...]

No unfortunately not, the only option I see is like I did, flashed some magic home bulbs with HAA and enabled the HomeKit Rest API to be able to do some extended smarts if needed.

pilot1981, 2020-11-26:
Ok, I created my philips hue developer account and now I'm able to power on and power off my HE lights using this HTTP requests:

```
$ curl -X PUT -d '{"on":true}' http://$ADDR/api/$HUE_USER/lights/1/state
```
that I could transform in this JSON?

{
  "c": { "l": 13 },
  "a": [{
    "t":24,
    "g":14,
    "n":4,
    "j":30,
    "y0": [{
      "v": 0,
      "r": 1,
      "0": { "h": [{ "h": "http://$ADDR/api/$HUE_USER/lights/1/state", "c": "{"on":true}" }] }
    }]
  }]
}

```
$ curl -X PUT -d '{"on":false}' http://$ADDR/api/$HUE_USER/lights/1/state
```
that could be this one:

{
  "c": { "l": 13 },
  "a": [{
    "t":24,
    "g":14,
    "n":4,
    "j":30,
    "y0": [{
      "v": 0,
      "r": 1,
      "0": { "h": [{ "h": "http://$ADDR/api/$HUE_USER/lights/1/state", "c": "{"on":true}" }] }
    }]
  }]
}

I understood well?

the JSON could works?

...«It – could – work!»...

pilot1981, 2020-11-26:
JSON validator give me an error on the "c" clause...what's it wrong?

pilot1981, 2020-11-26:
this one could be better ? :-)

{
"c": { "l": 13 },
"a": [{
"t":24,
"g":14,
"n":4,
"j":30,
"y0": [{
"v": 0,
"r": 1,
"0": { "h": [{ "h": "http://$ADDR/api/$HUE_USER/lights/1/state", "m": 1, "c": "{\"on\":true}" }] }
}]
}]
}

RavenSystem, 2020-11-27:
Replace $ADDR and $HUE_USER with normal values:

{
"c": { "l": 13 },
"a": [{
"t":24,
"g":14,
"n":4,
"j":30,
"y0": [{
"v": 0,
"r": 1,
"0": { "h": [{ "h": "$ADDR", "u": "api/$HUE_USER/lights/1/state", "m": 1, "c": "{\"on\":true}" }] }
}]
}]
}

pilot1981, 2020-11-27:
Hi @RavenSystem, thanks for suggestion, but if I check this JSON on validator:

{
"c": { "l": 13 },
"a": [{
"t":24,
"g":14,
"n":4,
"j":30,
"y0": [{
"v": 0,
"r": 1,
"0": { "h": [{ "h": "http://192.168.4.150/api/xxxxxxxxxxxxxxxxxxxP37rXVhQi/lights/1/state", "m": 1, "c": "{"on":false}" }] }
}]
}]
}

I obtain error, so I tried this version of JSON:

{
"c": { "l": 13 },
"a": [{
"t":24,
"g":14,
"n":4,
"j":30,
"y0": [{
"v": 0,
"r": 1,
"0": { "h": [{ "h": "http://192.168.4.150/api/xxxxxxxxxxxxx37rXVhQi/lights/1/state", "m": 1, "c": "{\"on\":false}" }] }
}]
}]
}

It passed the JSON check...

pay attention to this config: "{\"on\":false}"

so which version of JSON I have to use?

RavenSystem, 2020-11-27:
Try mine again. GitHub hides \\

xrust83, 2020-11-28:
{
  "c": {
    "l": 13,
    "n": "desk-lamp:temp/hum",
    "b": [
      {
        "g": 0,
        "t": 5
      }
    ]
  },
  "a": [
    {
      "0": {
        "r": [
          {
            "g": 12,
            "h": [
              {
                "h": "192.168.31.195",
                "u": "data?mes=off"
              }
            ]
          }
        ]
      },
      "1": {
        "r": [
          {
            "g": 12,
            "v": 1,
            "h": [
              {
                "h": "192.168.31.195",
                "u": "data?mes=on"
              }
            ]
          }
        ]
      },
      "b": [
        {
          "g": 0
        }
      ],
      "s": 0
    },
    {
      "t": 24,
      "b": [],
      "n": 4,
      "g": 14,
      "j": 90
    }
  ]
}

help me too please
when I go to http://192.168.31.195/data?mes=On on my watch, "On" is displayed, I tried to paste the code in which above, but nothing happens on the watch, although there is no error when saving the code

xrust83, 2020-11-28:
I realized my mistake

{
  "c": {
    "l": 13,
    "n": "desk-lamp:temp/hum",
    "b": [
      {
        "g": 0,
        "t": 5
      }
    ]
  },
  "a": [
    {
      "0": {
        "r": [
          {
            "g": 12
          }
        ],
        "h": [
          {
            "h": "192.168.31.195",
            "u": "data?mes=DESK%20LAMP:%20OFF"
          }
        ]
      },
      "1": {
        "r": [
          {
            "g": 12,
            "v": 1
          }
        ],
        "h": [
          {
            "h": "192.168.31.195",
            "u": "data?mes=DESK%20LAMP:%20ON"
          }
        ]
      },
      "b": [
        {
          "g": 0
        }
      ],
      "s": 0
    },
    {
      "t": 24,
      "b": [],
      "n": 4,
      "g": 14,
      "j": 90
    }
  ]
}

[https://www.youtube.com/watch?v=kkQxjYfrmdo](url)

pilot1981, 2020-12-09:
Ok, philips hue bulb arrived: I go to apply all in related world!

I'm able to change light build using these commands:

curl -X PUT -d '{"on":true}' http://192.168.4.150/api/LhHq--5tp9aBkiCZmoBnBg0yNyCr1qFP37rXVhQi/lights/2/state

curl -X PUT -d '{"on":false}' http://192.168.4.150/api/LhHq--5tp9aBkiCZmoBnBg0yNyCr1qFP37rXVhQi/lights/2/state

Now, I have this JSON to manage n.2 old switches/bulbs:

{"c":{"l":0,"i":1,"n":"lucebimbe","0":{"r":[{"g":16}]}},"a":[{"s":5,"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":13,"p":0},{"g":13,"t":0}]},{"s":5,"0":{"r":[{"g":15}]},"1":{"r":[{"g":15,"v":1}]},"b":[{"g":5,"p":0},{"g":5,"t":0}]},{"1":{"s":[{"a":1}]}},{"t":22,"h":0,"n":5,"j":5,"y0":[{"v":75,"r":1,"0":{"m":[{"g":1},{"g":2}]}}]}]}

and I thunk to change it to this JSON:

{"c":{"l":0,"i":1,"n":"lucebimbe","0":{"r":[{"g":16}]}},"a":[{"s":5,"0": { "h": [{ "h": "http://192.168.4.150/api/LhHq--5tp9aBkiCZmoBnBg0yNyCr1qFP37rXVhQi/lights/1/state", "m": 1, "c": "{\"on\":true}" },{"s":5,"0":{"r":[{"g":15}]},"1":{"r":[{"g":15,"v":1}]},"b":[{"g":5,"p":0},{"g":5,"t":0}]},{"1":{"s":[{"a":1}]}},{"t":22,"h":0,"n":5,"j":5,"y0":[{"v":75,"r":1,"0":{"m":[{"g":1},{"g":2}]}}]}]}}

but how I can send another HTTP command to shutdown lights?????

pilot1981, 2020-12-09:
I arrived to this JSON:

{"c":{"l":0,"i":1,"n":"lucebimbe","0":{"r":[{"g":16}]}},"a":[{"s":5,"0":{"h":[{"h":"http://192.168.4.150/api/LhHq--5tp9aBkiCZmoBnBg0yNyCr1qFP37rXVhQi/lights/1/state","m": 1,"c":"{\"on\":true}" }]},"1":{"h":[{"h":"http://192.168.4.150/api/LhHq--5tp9aBkiCZmoBnBg0yNyCr1qFP37rXVhQi/lights/1/state","m": 1,"c":"{\"on\":true}" }]}},{"s":5,"0":{"r":[{"g":15}]},"1":{"r":[{"g":15,"v":1}]},"b":[{"g":5,"p":0},{"g":5,"t":0}]},{"1":{"s":[{"a":1}]}},{"t":22,"h":0,"n":5,"j":5,"y0":[{"v":75,"r":1,"0":{"m":[{"g":1},{"g":2}]}}]}]}

but I don't understand how first accessory can understand to go to/from action 0 to/from action 1 related to GPIO input of wall switch...any suggestion?

pilot1981, 2020-12-10:
Hi,
I think I found answer to my question:

{"c":{"l":0,"i":1,"n":"lucebimbe","0":{"r":[{"g":16}]}},"a":[{"s":5,"0":{"h":[{"h":"http://192.168.4.150/api/LhHq--5tp9aBkiCZmoBnBg0yNyCr1qFP37rXVhQi/lights/1/state","m": 1,"c":"{\"on\":true}" }]},"1":{"h":[{"h":"http://192.168.4.150/api/LhHq--5tp9aBkiCZmoBnBg0yNyCr1qFP37rXVhQi/lights/1/state","m": 1,"c":"{\"on\":false}" }]},"b":[{"g":13,"p":0},{"g":13,"t":0}]},{"s":5,"0":{"r":[{"g":15}]},"1":{"r":[{"g":15,"v":1}]},"b":[{"g":5,"p":0},{"g":5,"t":0}]},{"1":{"s":[{"a":1}]}},{"t":22,"h":0,"n":5,"j":5,"y0":[{"v":75,"r":1,"0":{"m":[{"g":1},{"g":2}]}}]}]}

I go to test it in real world!

i3laze, 2020-12-12:
That’s great you found a solution to trigger your 3rd-party HomeKit device.

Just a note, that author doesn’t plan to implement inter-HAA notifications:
https://github.com/RavenSystem/esp-homekit-devices/issues/942

pilot1981, 2020-12-13:
@i3laze thanks for info, I think this isn't a problem when I go to control a light...

@RavenSystem I tested this JSON in real world but it's doesn't work (it doesn't power on philips hue lightbulb):

{"c":{"l":0,"i":1,"n":"lucebimbe","0":{"r":[{"g":16}]}},"a":[{"s":5,"0":{"h":[{"h":"http://192.168.4.150/api/LhHq--5tp9aBkiCxxxxxxxxqFP37rXVhQi/lights/1/state","m": 1,"c":"{\"on\":true}" }]},"1":{"h":[{"h":"http://192.168.4.150/api/LhHq--5tp9aBkiCxxxxxxxxqFP37rXVhQi/lights/1/state","m": 1,"c":"{\"on\":false}" }]},"b":[{"g":13,"p":0},{"g":13,"t":0}]},{"s":5,"0":{"r":[{"g":15}]},"1":{"r":[{"g":15,"v":1}]},"b":[{"g":5,"p":0},{"g":5,"t":0}]},{"1":{"s":[{"a":1}]}},{"t":22,"h":0,"n":5,"j":5,"y0":[{"v":75,"r":1,"0":{"m":[{"g":1},{"g":2}]}}]}]}

I also tried this one but HAA device don't accept JSON:

{"c":{"l":0,"i":1,"n":"lucebimbe","0":{"r":[{"g":16}]}},"a":[{"s":5,"0":{"h":[{"h":"http://192.168.4.150/api/LhHq--5tp9aBkiCxxxxxxxxqFP37rXVhQi/lights/1/state","m": 1,"c":"{"on":true}" }]},"1":{"h":[{"h":"http://192.168.4.150/api/LhHq--5tp9aBkiCxxxxxxxxqFP37rXVhQi/lights/1/state","m": 1,"c":"{"on":false}" }]},"b":[{"g":13,"p":0},{"g":13,"t":0}]},{"s":5,"0":{"r":[{"g":15}]},"1":{"r":[{"g":15,"v":1}]},"b":[{"g":5,"p":0},{"g":5,"t":0}]},{"1":{"s":[{"a":1}]}},{"t":22,"h":0,"n":5,"j":5,"y0":[{"v":75,"r":1,"0":{"m":[{"g":1},{"g":2}]}}]}]}

I tested HTTP command from my mac and it works fine...I have some other problem in my JSON?

pilot1981, 2020-12-15:
Hi @RavenSystem,
I read better your suggestions and now HTTP command works fine (thanks)...I have only last problem with physical button...I see using one time physical button and another time app apple home to switch on/off lightbulb I have some type of asynchronous behaviour...

I attached physical switch to GPIO 5 of shelly 2.5 (switch 2n) and this is my final JSON string:

{"c":{"l":0,"i":1,"n":"lucebimbe","0":{"r":[{"g":16}]}},"a":[{"s":5,"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":13,"p":0},{"g":13,"t":0}]},{"s":5,"0":{"h":[{"h":"192.168.4.150","u":"api/LhHq--5tp9aBkiCZmoBnBg0yNyCr1qFP37rXVhQi/lights/2/state","m": 1,"c":"{\"on\":false}" }]},"1":{"h":[{"h":"192.168.4.150","u":"api/LhHq--5tp9aBkiCZmoBnBg0yNyCr1qFP37rXVhQi/lights/2/state","m": 1,"c":"{\"on\":true}" }]},"b":[{"g":5,"p":0},{"g":5,"t":0}]},{"1":{"s":[{"a":1}]}},{"t":22,"h":0,"n":5,"j":5,"y0":[{"v":75,"r":1,"0":{"m":[{"g":1},{"g":2}]}}]}]}

I have to change B declaration or I have to use a stateless physical button?

pilot1981, 2020-12-16:
it works:

{"c":{"l":0,"i":1,"n":"lucebimbe","0":{"r":[{"g":16}]}},"a":[{"s":5,"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":13,"p":0},{"g":13,"t":0}]},{"s":5,"0":{"h":[{"h":"192.168.4.150","u":"api/LhHq--5tp9aBkiCZmoBnBg0yNyCr1qFP37rXVhQi/lights/2/state","m": 1,"c":"{\"on\":false}" }]},"1":{"h":[{"h":"192.168.4.150","u":"api/LhHq--5tp9aBkiCZmoBnBg0yNyCr1qFP37rXVhQi/lights/2/state","m": 1,"c":"{\"on\":true}" }]},"b":[{"g":5,"p":0},{"g":5,"t":0}]},{"1":{"s":[{"a":1}]}},{"t":22,"h":0,"n":5,"j":5,"y0":[{"v":75,"r":1,"0":{"m":[{"g":1},{"g":2}]}}]}]}

shelly 2.5 with button that send command philips hue light via HTTP, initial state of philips hue light setup in philips hue app, only little delay in answer...

final resul is:
- physical button on wall
- virtual button HAA without command of relay but only HTTP command (to use with automation)
- dummy switch to enter in setup mode
- virtual switch of philips hue to change command

also you obtain:
.- other physical button on wall
- virtual button HAA to command of relay

pilot1981, 2020-12-17:
this is JSON that don't send initial state via HTTP:

{"c":{"n":"HUE"},"a":[{"xa":0,"0":{"h":[{"h":"192.168.xxx.xxx","u":"api/<account>/lights/2/state","m": 1,"c":"{\"on\":false}" }]},"1":{"h":[{"h":"192.168.xxx.xxx","u":"api/<account>/lights/2/state","m": 1,"c":"{\"on\":true}" }]},"b":[{"g":5,"p":0},{"g":5,"t":0}]},{"1":{"s":[{"a":1}]}}]}

enhancement: I'm thinking could be interesting get philips hue lightbulb status using HTTP GET command and modify state of HAA switch in app apple home, it could be possible @Ravensystem?

pilot1981, 2020-12-21:
@RavenSystem it's possibile parse HTTP response and then do an action as change switch status?

RavenSystem, 2020-12-21:
> @RavenSystem it's possibile parse HTTP response and then do an action as change switch status?

No.

pilot1981, 2020-12-21:
@RavenSystem it's possibile to read status of another homekit device (es. philips hue lightbulb) and change HAA switch status?

I tried to use native homekit automation, but app apple home don't allow it!

RavenSystem, 2020-12-21:
> @RavenSystem it's possibile to read status of another homekit device (es. philips hue lightbulb) and change HAA switch status?

No.

> I tried to use native homekit automation, but app apple home don't allow it!

You should do it.

pilot1981, 2020-12-24:
Ok, I confirm final solution to create.2 homekit automation to sync status fo philips hue lightbulb and switch created by HAA on shelly

closed 2020-12-24 by pilot1981