RavenSystem/esp-homekit-devices #1215: MCP23017

paulzhdanov, 2020-11-16:
Does anybody have JSON examples for new MCP23017 support?

Tomfresse, 2020-12-08:
I have tested HAA support for MCP23017 i2c devices.
My testing JSON is:

{"c":{"z":0,"o":1,"ic":[{"c":5,"d":4,"q":400}],"mc":[{"b":0,"a":32,"ca":255}]},"a":[{"1":{"r":[{"g":100}]},"0":{"r":[{"g":100,"v":1}]} } ] }

Port A seems to work properly - except unnecessary port blinking at boot.

...then I started to test port B of MCP23017 ... and it seems not to work.

Here is the boot log:
```
Home Accessory Architect v3.8.5
(c) 2019-2020 José Antonio Jiménez Campos

NORMAL MODE

JSON:
 {"c":{"z":0,"o":1,"ic":[{"c":5,"d":4,"q":400}],"mc":[{"b":0,"a":32,"cb":255}]},"a":[{"1":{"r":[{"g":108}]},"0":{"r":[{"g":108,"v":1}]} } ] }

Hostname: HAA-260AD9
I2C bus: 0, scl: 5, sda: 4, freq: 400000, res: 0
MCP23017 0: bus: 0, addr: 32, chA: 0, chB: 255
Ping period: 4.9 secs
Setup mode time: 0 secs
Run HomeKit Server: 1
Unsecure connections: 0
mDNS TTL: 4500 secs
Gateway Ping: 255
Toggles to enter setup mode: 0
Exec Action: acc 0, action 0

ACCESSORY 1
Type 1
Serial: 260AD9-1
HAP v1.1.0
New GPIO Output 108 set to 1
New Digital Output Action 0: gpio 108, val 1, inch 0
New GPIO Output 108 set to 0
New Digital Output Action 1: gpio 108, val 0, inch 0
Exec Ac on Boot: 1
Init State: 0
Set init state
Setter ON
Exec Action: acc 1, action 0
Digital Output: gpio 108, val 1, inch 0
Free Heap: 40264

Device Category: 8
```

When switching the port ON and OFF in Homekit ... nothing happens.
I2C decoder shows the same value xFF send to PORT B:

```
Decoded I2C Protocol Result is:
Setup Write to [0x32] + ACK
0x01 + ACK
0xFF + ACK
Setup Write to [0x32] + ACK
0x01 + ACK
0xFF + ACK
```

In the same test PORT A of MCP23017 works fine:

```
Decoded I2C Protocol Result is
Setup Write to [0x32] + ACK
0x00 + ACK
0xFE + ACK
Setup Write to [0x32] + ACK
0x00 + ACK
0xFF + ACK_
```

I will continue testing ...

RavenSystem, 2020-12-08:
@Tomfresse There was a bug with channel B. It is fixed with last version v3.8.6.

closed 2020-12-08 by RavenSystem