RavenSystem/esp-homekit-devices #1216: BK7231U ?

TheReiner, 2020-11-17:
Has anyone ever successfully installed HAA firmware on a BEKEN chip?

[https://github.com/bekencorp/bk7231u_alios_sdk](url)

I received today a bulb from AliExpress but the chip is not an ESP8266 chip ... It looks like it would be very similar.

[image]

bj-tracer, 2020-11-17:
Hi!
Quote a source: "AliOS Things supports multiple architectures, including **ARM, C-Sky, MIPS, RISCV, rl78, rx600, xtensa** etc."

And what I found about this chip: "BK7231U is a 2.4 GHz 802.11n and **BLE 4.2** full feature comb chip with multi-media feature."

So it is not a esp8266 architecture.

TheReiner, 2020-11-17:
Why some have succeeded to install the firmware tasmota?

"BK7231U is a 2.4 GHz 802.11n and BLE 4.2 full feature comb chip with multi-media feature. It integrates hardware and software component to finish a complete 802.11b/g/n application, which supports AP and STA role simultaneously, and integrates Bluetooth Low Energy 4.2 transceiver and protocol stack. The 120 MHz MCU and embedded 256 KB RAM make it able to support multiple cloud links and also suitable for audio and image application.
Features
802.11 b/g/n 1x1 compliant
Wide-band and narrow-band interference detection and alleviation
20/40 MHz bandwidth and STBC
Working mode STA, AP, Direct
AP and STA role simultaneously
Classic Bluetooth co-existence
High-Speed 32-bit MCU，up to 120 MHz
256 KB internal data RAM
Dual I2C interface
Dual high speed UART with flow control capability
Microphone signal amplify
50 MHz SDIO interface and SPI
Interface with flow control
QSPI RAM direct mapping to data RAM space
Six 32-bit timer and one always on timer
Six 32-bit PWM with either high speed clock or low power clock
QSPI for RAM and FLASH extension
Full speed USB host and device
Multi-channel ADC with high speed 10-bits or low speed 16-bit with internal decimation filter
CMOS sensor 8-bit DVP
VGA Motion JPEG hardware encoder and dedicate DMA
32 bytes eFUSE and true random number generator
26 MHz and 32 KHz clock signal output"

tonysprenk, 2020-11-17:
Some manufacturers are using multiple architectures in the same product line. Or sometimes they use esp chips first and then switch over to other architectures. HAA and tasmota are certainly only compatible with esp chips.

closed 2020-12-17 by stale[bot]