RavenSystem/esp-homekit-devices #1217: SonOff Basic - Help needed with initial status and notifications

mpvaladares, 2020-11-17:
Hi,
I am trying to make my SonOff Basic return from the previous status after a power outage, but it isn't working as expected.
I am using the following JSON:
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":1,"s":5,"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}]}]}

Also, I was wondering how I can set it up to receive a notification on my Apple devices when the switch is turned (on or off). I can turn the switch using HomeKit, but I don't receive notifications. Is it possible?

Thanks in avance,
mpvaladares

rbrtsluk, 2020-11-18:
I get notifications with the garage json so maybe check that out

juliusbangert, 2020-11-21:
Hi @rbrtsluk . Please could you point me in the direction of the garage json you mentioned here?
I'm trying to get a Sonoff basic r3 working and need a json file for this. Thanks.

rbrtsluk, 2020-11-21:
This was the json I used {"c":{"l":13,"b":[{"g":0,"t":5}],"0":{"r":[{"g":12}]}},"a":[{"t":40,"d":18,"f3":[{"g":14,"t":0}],"f4":[{"g":14}],"0":{"r":[{"g":12,"v":1,"i":0.5}]},"1":{"a":0},"2":{"r":[{"g":12,"i":1.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},"3":{"a":2}}]}

closed 2020-12-22 by stale[bot]