RavenSystem/esp-homekit-devices #1219: MagicHome RGB Controller help!!!

epicsepi, 2020-11-20:
I have two MagicHome RGB controller, but I got 4 swiches in Home app: The two standard rgb switch + two single on/off switch (doesn’t works) I can’t delete this two unnecessary switch. I think thease are the bridges or something?!...?!!!

I think JSON is the problem. Current JSON:

{"c":{"n":"MagicHomeRGB"},"a":[{"t":30,"r":5,"g":12,"v":13},{"1":{"s":[{"a":1}]},"i":0.5}]}
[image]

Please help me how to delete these.

pilot1981, 2020-11-21:
Perhaps, they Are dummy switch to enter in setup mode?! Please group them in a separate room to Maintanance purpose!

Kristian8606, 2020-11-21:
Remove it from JSON and it will disappear

epicsepi, 2020-11-21:
> Remove it from JSON and it will disappear

Ok, but how? What I have to remove from Json?

Kristian8606, 2020-11-24:
Try this code
`{
  "c": 
  {
    "n": "MagicHomeRGB"
  },
  "a": 
  {
    "t": 30,
    "r": 5,
    "g": 12,
    "v": 13
  }
} `

Kristian8606, 2020-11-24:
I'm not sure if it works since I edited it over the phone

wayn3, 2020-11-24:
### With setup switch (current)
```
{
	"c":{
		"n":"MagicHomeRGB"
	},
	"a":[{
		"t":30,
		"r":5,
		"g":12,
		"v":13
	},{
		"1":{"s":[{"a":1}]},
		"i":0.5
	}]
}
```
### without setup switch
```
{
	"c":{
		"n":"MagicHomeRGB"
	},
	"a":[{
		"t":30,
		"r":5,
		"g":12,
		"v":13
	}]
}
```

closed 2020-12-25 by stale[bot]