RavenSystem/esp-homekit-devices #1221: stepper motor documentation

Izac9, 2020-11-21:
can you please upload the documentation for stepper motors? how to make JSON? Some example best using 4 GPIOs. Thank you so much.

Dylanwhat, 2020-11-22:
Yes this would be very helpful for me as well. I love HAA and have used it for several projects, but I have had to use other projects (The Smart Home Hookup, nidayand) to set up my smart blinds because I don't understand how to drive a 28BYJ-48 with HAA. I asked in the Discord and it seemed that HAA only supports running a motor from a relay, such as a Sonoff Dual. I'm a bit frustrated with the 28BYJ-48 anyway because it doesn't have the torque required to completely close the blinds. My ideal solution would be a more powerful stepper with HAA. What are you hoping to drive, @Izac9?

Izac9, 2020-11-22:
> Yes this would be very helpful for me as well. I love HAA and have used it for several projects, but I have had to use other projects (The Smart Home Hookup, nidayand) to set up my smart blinds [...]

I have 28byj-48 too with ULN2003 and I have Roller blinds that I want to be smart 😁😁

closed 2021-01-13 by stale[bot]