RavenSystem/esp-homekit-devices #1222: help for Wemos d1 mini - pin connection

pasparakis, 2020-11-23:
hello everyone... first flashing my Wemos d1 mini the (Tasmota.bin) and in the second time flashing the (fullhaaboot.bin).I'm doing a project to check an air conditioner Fujitsu, and I used the json from devices database : 

Fujitsu HVAC Ir Remote Thermostat and Humidity sensor | 3: IrDA LED, 14: DHT22 sensor | {"c":{"t":3},"a":[{"t":25,"w":3,"g":14,"h":-3,"m":19,"x":28,"ds":0.5,"df":0.5,"p":"H0C#AF0:AFB)AF","0":{"i":[{"c":"bAaAcBcBmAgAdAfAaF"}]},"1":{"i":[{"c":"bAaAcBcBmAgAdHbAdAcBbAdAaAbAgAzhAcBe"}]},"2":{"i":[{"c":"bAaAcBcBmAgAdHbAdAcBbAbBaAaAiAzhAbAcAaAa"}]},"3":{"i":[{"c":"bAaAcBcBmAgAdHbAdAcBbAdAaAbAeBziAbCe"}]},"4":{"i":[{"c":"bAaAcBcBmAgAdHbAdAcBbAbBaAaAgBziAcAbAaAa"}]},"8":{"i":[{"c":"bAaAcBcBmAgAdHbAdAcBbAbAbAcAgAzhAcDaAa"}]},"9":{"i":[{"c":"bAaAcBcBmAgAdHbAdAcBbAcBaBiAzhAbAbE"}]},"10":{"a":1},"11":{"a":2},"5":{"m":[{"g":1,"v":0.02}]}}]}
-- | -- | --

the pin I used on Wemos is for dht22 is (D5-GPIO14) and works properly ..

the pin I used on Wemos is for ir transmitter is (RX-GPIO3) but it does not work

Any ideas?

pasparakis, 2020-11-25:
any help please..

closed 2020-11-28 by pasparakis

xrust83, 2020-12-04:
Are you sure the gpio3 port is not working? look at the ir-led through the older model mobile phone camera

pasparakis, 2020-12-04:
Yes I'm sure but and the protocol is wrong "p":"H0C#AF0:AFB)AF" because my remote control is different..

xrust83, 2020-12-04:
you need one more vemos v1 mini and an IR receiver, assemble the following design

download the HAA IR Capture Tool app and get your remote codes. Next, from the code in the screenshot, create your own codes for each button, this is from 15 to 45 minutes of your time
https://github.com/RavenSystem/esp-homekit-devices/wiki/HAA-Infra-Red#raw-mode
https://github.com/RavenSystem/esp-homekit-devices/wiki/HAA-IR-Encoder-Table

это не сложно но нужно быть внималельным
[image]

pasparakis, 2020-12-04:
Οοοο!!!!fried I will try this option ...thank you... if I have difficulty I will tell you

pasparakis, 2020-12-07:
your help was invaluable & thank you..but I have one last problem..I do not know what values ​​to find from the remote control to the thermostat actions ..I will send you a photo ..[image]

pasparakis, 2020-12-07:
for example: what temperatures and what fan speeds?

xrust83, 2020-12-17:
As far as I understand, you must set the temperature and speed on your remote control from the air conditioner. Intercept it with a photo tool, and then make a short code from the intercepted code as in the examples above. Explore the topic deeper. I only taught TV

pasparakis, 2020-12-23:
thanks for your time friend..