RavenSystem/esp-homekit-devices #1223: problem with mDNS and AP Eero

pilot1981, 2020-11-24:
Hi,
I using HAA version 3.7.2.

I'm using "n" option on JSON to config hostname via mDNS, all works fine when AP Eero were configured in bridge mode and DNS server and DHCP server was the Vodafone Power Station.

There is an example of JSON I'm using on my devices:

{"c":{"n":"lucebalcone"},"a":[{"s":5,"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":5,"p":0},{"g":5,"t":0}]},{"i":0.5,"1":{"s":[{"a":1}]}}]}

Now, I configured Ap  Eero to act as gateway/router and mDNS stop working..I obtained this output from cli of my laptop:

MacBook-Pro-di-daniele-147:~ danielevelardi$ dns-sd -B _hap._tcp local
Browsing for _hap._tcp.local
DATE: ---Tue 24 Nov 2020---
10:02:20.650  ...STARTING...
Timestamp     A/R    Flags  if Domain               Service Type         Instance Name
10:02:21.101  Add        3   4 local.               _hap._tcp.           Netatmo Presence
10:02:21.101  Add        3   4 local.               _hap._tcp.           Netatmo Welcome
10:02:21.101  Add        3   4 local.               _hap._tcp.           Netatmo Giardino
10:02:21.101  Add        3   4 local.               _hap._tcp.           VELUX Gateway
10:02:21.101  Add        3   4 local.               _hap._tcp.           HAA-A995E3
10:02:21.101  Add        3   4 local.               _hap._tcp.           HAA-2470C0
10:02:21.101  Add        3   4 local.               _hap._tcp.           HAA-76F77B
10:02:21.101  Add        3   4 local.               _hap._tcp.           Aqara Hub-A8D7
10:02:21.101  Add        3   4 local.               _hap._tcp.           Sala da pranzo eero
10:02:21.101  Add        3   4 local.               _hap._tcp.           TRADFRI gateway
10:02:21.101  Add        3   4 local.               _hap._tcp.           HAA-05CF3A
10:02:21.101  Add        3   4 local.               _hap._tcp.           HAA-9670AA
10:02:21.101  Add        3   4 local.               _hap._tcp.           HAA-93C08E
10:02:21.101  Add        3   4 local.               _hap._tcp.           HAA-52077E
10:02:21.101  Add        3   4 local.               _hap._tcp.           HAA-7402AC
10:02:21.101  Add        3   4 local.               _hap._tcp.           HAA-24C991
10:02:21.101  Add        3   4 local.               _hap._tcp.           HAA-745843
10:02:21.101  Add        3   4 local.               _hap._tcp.           HAA-59DD5F
10:02:21.101  Add        3   4 local.               _hap._tcp.           HAA-93BD51
10:02:21.101  Add        3   4 local.               _hap._tcp.           HAA-2466B6
10:02:21.101  Add        3   4 local.               _hap._tcp.           HAA-93B742
10:02:21.101  Add        3   4 local.               _hap._tcp.           HAA-2C7166
10:02:21.101  Add        3   4 local.               _hap._tcp.           HAA-8D94D7
10:02:21.101  Add        3   4 local.               _hap._tcp.           HAA-7453D9
10:02:21.101  Add        3   4 local.               _hap._tcp.           HAA-8CD26E
10:02:21.102  Add        2   4 local.               _hap._tcp.           HAA-93BEA9
10:02:21.890  Add        2   4 local.               _hap._tcp.           telecamera sala da pranzo
10:02:38.970  Rmv        0   4 local.               _hap._tcp.           HAA-2470C0
10:02:50.112  Rmv        0   4 local.               _hap._tcp.           HAA-05CF3A
10:03:03.782  Add        2   4 local.               _hap._tcp.           HAA-05CF3A

I see only default hostname, but not hostname sent via mDNS...

RavenSystem, 2020-11-24:
`"n":` sets device hostname to help user to identify it (with compatible routers), but not mDNS name (internal HomeKit name). mDNS name is always HAA-XXXXXX, because it is short, to use low memory, and it is different for each accessory, to avoid conflicts.

RavenSystem, 2020-11-24:
About AP Eero used as gateway, I don't know where is the problem. As you can see in your MacBook terminal, HomeKit devices are announcing mDNS right. Maybe you must tune your router, enabling IGMP snooping to manage multicast traffic, or maybe other settings. I don't know that router.

pilot1981, 2020-11-24:
MacBook-Pro-di-daniele-147:~ danielevelardi$ ping HAA-05CF3A.local
PING haa-05cf3a.local (192.168.4.50): 56 data bytes
Request timeout for icmp_seq 0
64 bytes from 192.168.4.50: icmp_seq=0 ttl=255 time=1086.352 ms
64 bytes from 192.168.4.50: icmp_seq=1 ttl=255 time=96.391 ms
64 bytes from 192.168.4.50: icmp_seq=2 ttl=255 time=25.419 ms

but not ping lucebimbe

RavenSystem, 2020-11-24:
HAA-XXXXXX.local is mDNS name, and its resolution is provided by mDNS environment, not by your router.

To use hostnames, you need a router compatible with DHCP and DNS hostname resolution names.

pilot1981, 2020-11-24:
I think Eero have only DHCP server, so if it don't have DNS server...I can solve DNS name...I understood?

pilot1981, 2020-11-24:
Eero support confirmed IGMP snooping isn't supported!

RavenSystem, 2020-11-24:
No DNS server, no IGMP snooping... it doesn't seem a good device.

pilot1981, 2020-11-24:
In your opinion, if I setup a new DNS server in the same network, in your opinion can it work? or IGMP snooping is mandatory?

RavenSystem, 2020-11-24:
IGMP snooping is not mandatory, but it is very useful when there are a lot of multicast traffic (HomeKit, AirPlay, DLNA...).

You must setup a new DNS server and a new DHCP server. Then, your DHCP server must register local network names on new local DNS server. And all your devices must use that new DNS server.

pilot1981, 2020-11-24:
I will use DHCP server of Eero and setup it to release DNS server IP of new DNS server I will add in the same network...I try...

wayn3, 2020-11-24:
I seem to get hostname through fine
[image]

pilot1981, 2020-11-26:
yes but @wayn3 you are able to do a ping to the hostname via cli?

pilot1981, 2020-11-26:
I tried to use DHCP server eero and DNS server on HA on the same network but it don't solve my problem...

pilot1981, 2020-12-14:
the curious thing is that I'm able to see hostname written in HAA from EERO APP, but if I usa CLI of may MacOS using ping command...it don't is able to resolve hostname...very strange

pilot1981, 2020-12-21:
Eero AP is able to read hostname of HAA device and use it as nickname of device, but it don't use it as hostname...

So I aksed a new feature request to Eero support to import it into DNS server to Eero AP...

closed 2020-12-21 by pilot1981