RavenSystem/esp-homekit-devices #1226: Garage door JSON - help

jakudo, 2020-11-25:
Hi,

I have Sonoff SV with 2 sensors (open and close) flashed as Garage door. It work great, however, when I need to enter the setup mode (so take the power off then on and within 3 seconds off again and on again) it opens the garage door somehow. I would like to have it closed (this night, there was some power outages and the power "blinked" so fast that the sonoff went to setup mode and open the door at around 2 am).

Here is my JSON
`{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"t":40,"d":23,"f3":[{"g":14,"t":0}],"f4":[{"g":14}],"f2":[{"g":4,"t":0}],"f5":[{"g":4}],"0":{"r":[{"g":12,"v":1,"i":0.7}]},"1":{"a":0},"2":{"r":[{"g":12,"i":0.7},{"g":12,"v":1,"i":0.7},{"g":12,"v":1,"i":0.7}]},"3":{"a":2}}]}`

What can I modify/add/delete so it doesn't create the impulse after short power outage and so the door would stay closed?

Thanks,
Jakub

macjl, 2020-11-25:
Hello,
Add  `"0":{"r":[{"g":12}]}` in the config options :

`{"c":{"l":13,"b":[{"g":0,"t":5}],"0":{"r":[{"g":12}]}},"a":[{"t":40,"d":23,"f3":[{"g":14,"t":0}],"f4":[{"g":14}],"f2":[{"g":4,"t":0}],"f5":[{"g":4}],"0":{"r":[{"g":12,"v":1,"i":0.7}]},"1":{"a":0},"2":{"r":[{"g":12,"i":0.7},{"g":12,"v":1,"i":0.7},{"g":12,"v":1,"i":0.7}]},"3":{"a":2}}]}`

jakudo, 2020-11-25:
Awesome! Thanks a lot. It works. I don't really understand how (I thought that you cannot use action twice in one JSON), but it works. Really thank you 👍

closed 2020-11-25 by jakudo