RavenSystem/esp-homekit-devices #1228: Sonoff TH LED blue

pixidixi88, 2020-11-25:
Hello,
I would like to use blue LED in my Sonoff TH and present it in the home app as standard switch. It doesn't work. What is wrong with below json.

Thank You!

{
  "c": {
    "b": [
      {
        "g": 0,
        "t": 5
      }
    ]
  },
  "a": [
    {
      "0": {
        "r": [
          {
            "g": 12
          }
        ]
      },
      "1": {
        "r": [
          {
            "g": 12,
            "v": 1
          }
        ]
      },
      "b": [],
      "s": 0
    },
    {
      "0": {
        "r": [
          {
            "g": 13,
            "v": 1
          }
        ]
      },
      "1": {
        "r": [
          {
            "g": 13,
            "v": 0
          }
        ]
      },
      "t": 1,
      "b": [],
      "s": 0,
      "f0": []
    },
    {
      "t": 3,
      "b": [],
      "s": 0,
      "f0": [
        {
          "g": 0,
          "t": 1
        }
      ],
      "f1": [
        {
          "g": 0,
          "t": 2
        }
      ],
      "f2": [
        {
          "g": 0,
          "t": 3
        }
      ]
    }
  ]
}

xrust83, 2020-11-28:
{"c":{"b":[{"g":0,"t":5}],"l":13},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[],"s":0},{"0":{" r ":[{" g ":13," v ":1}]}," 1 ":{" r ":[{"g":13,"v":0}]},"t":1,"b":[],"s":0,"f0":[]},{"t":3,"b":[],"s":0,"f0":[{"g":0,"t":1}],"f1":[{"g":0,"t":2}],"f2":[{"g":0,"t":3}]}]}

https://glumb.github.io/haa-configurator/

pixidixi88, 2020-12-02:
Yes, I did it in the configurator and also copy paste from above post but still it doesn't work.

xrust83, 2020-12-02:
{"c":{"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12},{"g":13,"v":0}]},"1":{"r":[{"g":12,"v":1},{"g":13,"v":1}]},"b":[],"s":0},{"t":3,"b":[],"s":0,"f0":[{"g":0,"t":1}],"f1":[{"g":0,"t":2}],"f2":[{"g":0,"t":3}]}]}

closed 2021-01-01 by stale[bot]