RavenSystem/esp-homekit-devices #1229: Unable to flash on Mac OS BigSur

tlagarto, 2020-11-26:
I'm far from proficient on this but I was able to flash quite a few of devices in the last months (following all the instructions and tutorials available).
Now, after update OS to BigSur, I can't even erase the original firmware.
All I get back in terminal is:
`Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.8/bin/esptool.py", line 57, in <module>
    import serial.tools.list_ports as list_ports
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/serial/tools/list_ports.py", line 29, in <module>
    from serial.tools.list_ports_posix import comports
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/serial/tools/list_ports_posix.py", line 31, in <module>
    from serial.tools.list_ports_osx import comports
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/serial/tools/list_ports_osx.py", line 32, in <module>
    kIOMasterPortDefault = ctypes.c_void_p.in_dll(iokit, "kIOMasterPortDefault")
ValueError: dlsym(RTLD_DEFAULT, kIOMasterPortDefault): symbol not found
`

Does any one know a fix or workaround?

rjblake, 2020-11-26:
Perhaps try using Node MCU PyFlasher - you can get a Mac binary [here](https://github.com/marcelstoer/nodemcu-pyflasher/releases)

Kristian8606, 2020-11-27:
I experienced the same problem and it helped me.
 https://github.com/espressif/esptool/issues/540#issuecomment-650507813

TheReiner, 2020-11-28:
pip3 install pyserial --pre

There is an issue here ;) 
[https://github.com/pyserial/pyserial/issues/509](url)

tlagarto, 2020-12-02:
> I experienced the same problem and it helped me.
> [espressif/esptool#540 (comment)](https://github.com/espressif/esptool/issues/540#issuecomment-650507813)

I've tried this one (since it seemed to be the easiest one for my skill level) and it worked!
Thank you

closed 2021-01-01 by stale[bot]