RavenSystem/esp-homekit-devices #1230: help for Fujitsu remote ac - HomeKit

pasparakis, 2020-11-27:
hi, I am a beginner..can anyone help me to make a controller for Fujitsu air-conditioner to work with HomeKit?
I need more in which order to flash and which files .. 
ps: use wemos d1 mini - dht 22 - ir transmitter..thank you for your time ..

scout1256, 2020-11-27:
HAA IR didn't work 4 me. so i mad arduino ide ir raw collector. using excel to configure raw data to haa compatible data. then install haa on esp01s. done.

sorry 4 this typing, one hand only

pasparakis, 2020-11-27:
thank you my friend..first flashing my Wemos d1 mini the (Tasmota.bin) and in the second time flashing the (fullhaaboot.bin).I'm doing a project to check an air conditioner Fujitsu, and I used the json from devices database :  

is correctly steps?
and second step I placed this  json from database :  

  {"c":{"t":3},"a":
[{"t":25,"w":3,"g":14,"h":-3,"m":19,"x":28,"ds":0.5,"df":0.5,"p":"H0C#AF0:AFB)AF","0":{"i":[{"c":"bAaAcBcBmAgAdAfAaF"}]},"1":{"i":[{"c":"bAaAcBcBmAgAdHbAdAcBbAdAaAbAgAzhAcBe"}]},"2":{"i":[{"c":"bAaAcBcBmAgAdHbAdAcBbAbBaAaAiAzhAbAcAaAa"}]},"3":{"i":[{"c":"bAaAcBcBmAgAdHbAdAcBbAdAaAbAeBziAbCe"}]},"4":{"i":[{"c":"bAaAcBcBmAgAdHbAdAcBbAbBaAaAgBziAcAbAaAa"}]},"8":{"i":[{"c":"bAaAcBcBmAgAdHbAdAcBbAbAbAcAgAzhAcDaAa"}]},"9":{"i":[{"c":"bAaAcBcBmAgAdHbAdAcBbAcBaBiAzhAbAbE"}]},"10":{"a":1},"11":{"a":2},"5":{"m":[{"g":1,"v":0.02}]}}]}

everything works components except transmitter..while transmitting a signal (check with camera from mobile)
the aircondition is dead..

the transimitter that I use is :  

[image]

thank you for your time..

peros550, 2020-11-27:
This transmitter worked for me in the past but the distance wasn't great. You might have better luck with a circuit which includes a transistor so the IR signal can be amplified.

pasparakis, 2020-11-27:
to me it does not work at all ..
are the steps I did correct?

peros550, 2020-11-27:
You didn't have to install tasmota.bin but nevertheless I understand you have HAA installed in your Wemos D1 already. So that's good. 
The mobile phones (especially iphones) have a filter in their camera so if you are not seeing the IR blaster it doesn't mean the IR is not working. 

What pin did you use for connecting the DATA pin of your IR module? D5?

pasparakis, 2020-11-27:
I saw that it works properly with a different Samsung module..I'm sure it works ..

the pin where I connected the data is gpio 3 (rx) on d1 module ..

so it says in the database ..

3: IrDA LED, 14: DHT22 sensor

peros550, 2020-11-27:
ok, sorry my bad on the pin. If still not sure about the AC json, I would propose to try with a TV json for your TV manufacturer. If that works, then your module works fine. If it doesn't work with a TV json, your module does not produce strong enough light.

pasparakis, 2020-11-27:
Thanks fried for your time... I hope to fix it...

closed 2020-11-28 by pasparakis